jpnurmi/flutter_libserialport #3: Example or tutorial needed

dariocavada, 2021-01-02:
First of all thank you for this package.

Would like to know if it's possible to have an example for reading data from a serial port using flutter.

My goal is to use the serial port on an Android Set Top Box (N1 Max) capable of reading data from Arduino.

jpnurmi, 2021-01-05:
Yes, you are absolutely right that there should be better examples. Do you have a suggestion for what the examples could read and write? The typical problem with serial port examples - should one assume dummy echoes, a certain protocol, or something else? :) Anyway, one option could be to adapt/port the dummy examples from libserialport.

dariocavada, 2021-01-15:
This is a Flutter component, so some example with widgets and business logic would be very useful:

A typical read and write from/to Arduino for exampe...
And eventually a complete example (in the example folder) that shows how:
1. Configure the port (velocity, parity, ...)
2. Select the port from a dropdown
3. A button to connect 
3. If success start to listen to the serial port, if you receive a string show the string in a list 
4. If error, shows an error (and eventually retry to connect after a while)
5. a text field + a button to send a string to Arduino

rty813, 2021-02-03:
@dariocavada You could see this site for more examples. Its same.
[https://pub.dev/packages/dart_serial_port](https://pub.dev/packages/dart_serial_port)