jpnurmi/flutter_libserialport #4: Permission Denied on Android Example

EParisot, 2021-01-05:
Hi I cloned the repo, jumped in /examples and executed a "flutter run" command...
The app launched, I then connected an ESP32 to USB port on my phone and tapped on the refresh button.

This happened :

`W/1.ui    ( 5446): type=1400 audit(0.0:22048): avc: denied { read } for name="tty" dev="sysfs" ino=18303 scontext=u:r:untrusted_app:s0:c98,c257,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0      
I/flutter ( 5446): ══╡ EXCEPTION CAUGHT BY GESTURE ╞═══════════════════════════════════════════════════════════════════
I/flutter ( 5446): The following SerialPortError was thrown while handling a gesture:
I/flutter ( 5446): Permission denied, errno = 13
I/flutter ( 5446):
I/flutter ( 5446): When the exception was thrown, this was the stack:
I/flutter ( 5446): #0      Util.call (package:dart_serial_port/src/util.dart:39:7)
I/flutter ( 5446): #1      _SerialPortImpl.availablePorts (package:dart_serial_port/src/port.dart:230:21) 
I/flutter ( 5446): #2      SerialPort.availablePorts (package:dart_serial_port/src/port.dart:82:61)       
I/flutter ( 5446): #3      _ExampleAppState.initPorts.<anonymous closure> (package:flutter_serial_port_example/main.dart:38:48)
I/flutter ( 5446): #4      State.setState (package:flutter/src/widgets/framework.dart:1268:30)
I/flutter ( 5446): #5      _ExampleAppState.initPorts (package:flutter_serial_port_example/main.dart:38:5)I/flutter ( 5446): #6      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:990:20)
I/flutter ( 5446): #7      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:183:24)
I/flutter ( 5446): #8      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:598:11)I/flutter ( 5446): #9      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:287:5)I/flutter ( 5446): #10     BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:222:7)
I/flutter ( 5446): #11     PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:476:9)
I/flutter ( 5446): #12     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:93:12)
I/flutter ( 5446): #13     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:138:9)
I/flutter ( 5446): #14     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:387:8)
I/flutter ( 5446): #15     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:136:18)
I/flutter ( 5446): #16     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:122:7)   
I/flutter ( 5446): #17     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:381:19)  
I/flutter ( 5446): #18     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:361:22)I/flutter ( 5446): #19     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:280:11)
I/flutter ( 5446): #20     GestureBinding._handlePointerEventImmediately (package:flutter/src/gestures/binding.dart:316:7)
I/flutter ( 5446): #21     GestureBinding.handlePointerEvent (package:flutter/src/gestures/binding.dart:280:5)
I/flutter ( 5446): #22     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:238:7)
I/flutter ( 5446): #23     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:221:7)
I/flutter ( 5446): #27     _invoke1 (dart:ui/hooks.dart:176:10)
I/flutter ( 5446): #28     PlatformDispatcher._dispatchPointerDataPacket (dart:ui/platform_dispatcher.dart:276:7)
I/flutter ( 5446): #29     _dispatchPointerDataPacket (dart:ui/hooks.dart:96:31)
I/flutter ( 5446): (elided 3 frames from dart:async)
I/flutter ( 5446):
I/flutter ( 5446): Handler: "onTap"
I/flutter ( 5446): Recognizer:
I/flutter ( 5446):   TapGestureRecognizer#77b4e
I/flutter ( 5446): ════════════════════════════════════════════════════════════════════════════════════════════════════
W/1.ui    ( 5446): type=1400 audit(0.0:22050): avc: denied { read } for name="tty" dev="sysfs" ino=18303 scontext=u:r:untrusted_app:s0:c98,c257,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0      
I/flutter ( 5446): Another exception was thrown: SerialPortError: Permission denied, errno = 13
D/ViewRootImpl[MainActivity]( 5446): windowFocusChanged hasFocus=false inTouchMode=true
D/ViewRootImpl[MainActivity]( 5446): windowFocusChanged hasFocus=true inTouchMode=true
I/Surface ( 5446): opservice is null false
W/1.ui    ( 5446): type=1400 audit(0.0:22054): avc: denied { read } for name="tty" dev="sysfs" ino=18303 scontext=u:r:untrusted_app:s0:c98,c257,c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0      
I/flutter ( 5446): Another exception was thrown: SerialPortError: Permission denied, errno = 13
I/Surface ( 5446): opservice is null false`

I don't know about an Usb Serial permission in Android... how to solve this problem ?

Thanks in advance !

EParisot, 2021-01-05:
Hi and thank you.
Unfortunately, it seems the discussion you mentioned doesn't really deals with Android permissions... At least, it seems that a rooted device have no problem, but it's not my case. I can access to external memory of the phone with permission but /tty reside in the internal memory...

jpnurmi, 2021-01-05:
Sorry, I don't know how to deal with a non-rooted device.

EParisot, 2021-01-05:
> Sorry, I don't know how to deal with a non-rooted device.

Ok I'll try to figure it out... It may be possible since the usb_serial package works but I wanted to add Desktop compatibility to my app...
Thanks !

realrk69, 2021-01-31:
@EParisot Glad to see you here too.  I have not faced this issue on my non-rooted android. Maybe it's the device you're using?

EParisot, 2021-01-31:
Haha are we working on the same thing ?

How really ? That's interesting...
I use an ESP32 plugged in a OnePlus5... Android 10
What is your Android version ?

realrk69, 2021-01-31:
I'm working on a home environment monitoring project which auto triggers ac/heater accordingly to maintain 24 degrees C.
Am using ATmega custom-designed PCBs due to significant low cost and efficiency). My phone is Xiaomi Mi 9T/K20 Pro on Android 10. I don't like some aspects of this library that have to be made manually. Had to abstract a class with a reactive state just to detect if the device is plugged in or not. Would prefer if someone made a library or at least templates that allow streaming data through providers to change UI accordingly.

EParisot, 2021-01-31:
Ok looks cool project !
I don't really get why you need usb serial thought... But I ended with another package with a more complete API, as you know...

wrjcs, 2021-06-19:
> Ok looks cool project !
> I don't really get why you need usb serial thought... But I ended with another package with a more complete API, as you know...

Hello,
did you resolve this issue?

EParisot, 2021-06-19:
No I used usb_serial instead

SathyaPratapRudra, 2022-08-09:
hi all, even I am facing the permission issue with non rooted phone @realrk95 can you please share your sample code.

lucafabbri, 2024-02-21:
We put permission handling in todo. 
Activity is tracked in #76 
Closed

closed 2024-02-21 by lucafabbri