jpnurmi/flutter_libserialport #8: SerialPortReader

pouryaomidi, 2021-02-17:
Hi,

I am using your library to communicate with a serial device. I can write to the serial port, but I am only able to read from the serial port using the read() method. I can't use the SerialPortReader() to perform the same action. This is how I use it: 
`  /// read when response is enabled
  Future<String> readSerial() async {
     // _sp is the the serial port I am working with. 
    // This works
    print('num bytes available: ${(_sp.bytesAvailable)}');
    var data = _sp.read(_sp.bytesAvailable);
    print('data is ${utf8.decode(data)}');
    return utf8.decode(data);
    
    // This doesn't work
    log.info('reading from the serial port');
    final reader = SerialPortReader(_sp);
    if (reader != null) {
      log.info('listening to serial port');
      reader.stream.listen((data) {
        log.info('received: $data');
        return data.toString();
      });
      reader.close();
    }
    return null;
  }
}`

Any help is appreciated. 

P.S. It would be nice to mention in your example that the SerialPortConfig should be configured before opening the serial port. I assumed a default configuration was used and it took me a while to figure out that this should've been done. Thanks!

closed 2021-02-18 by pouryaomidi

ext-danny, 2022-04-22:
I'm unable to stream the data too. I had followed these steps to 

> Hi,
> 
> I am using your library to communicate with a serial device. I can write to the serial port, but I am only able to read from the serial port using the read() method. I can't use the SerialPortReader() to perform the same action. This is how I use it:
> 
> ```
>  /// read when response is enabled
>   Future<String> readSerial() async {
>      // _sp is the the serial port I am working with. 
>     // This works
>     print('num bytes available: ${(_sp.bytesAvailable)}');
>     var data = _sp.read(_sp.bytesAvailable);
>     print('data is ${utf8.decode(data)}');
>     return utf8.decode(data);
>     
>     // This doesnot work
>     log.info('reading from the serial port');
>     final reader = SerialPortReader(_sp);
>     if (reader != null) {
>       log.info('listening to serial port');
>       reader.stream.listen((data) {
>         log.info('received: $data');
>         return data.toString();
>       });
>       reader.close();
>     }
>     return null;
>   }
> }
> ```
> 
> Any help is appreciated.
> 
> P.S. It would be nice to mention in your example that the SerialPortConfig should be configured before opening the serial port. I assumed a default configuration was used and it took me a while to figure out that this should've been done. Thanks!

I'm unable to stream data either. I defined the ports, configured them and opened to read, but the streamer doesn't seem to provide any kind of data. Is there a reason this happens?

lastant, 2024-03-21:
Maybe it's because you call `reader.close();` right away?