jpnurmi/flutter_libserialport #11: stdlib.h and string.h not found

digitalkirkley, 2021-03-16:
Added the package as a plugin.  When launching debug on Linux Ubuntu (20.), it is unable to find stdlib.h within linux_termios.c and string.h within libserialport_internal.h.  Would you have any ideas what this might be please?

jpnurmi, 2021-03-16:
Do you have the `build-essential` package installed? Is the Linux section in `flutter doctor -v` all green? Could you run `flutter build linux -v` and paste the output after the red part (starting from somewhere "Building Linux application...")?

jpnurmi, 2021-03-16:
I have a fresh clean Ubuntu 20.04.1 LTS running in a Docker container. I cloned the latest Flutter and installed all packages suggested by `flutter doctor -v`. Here's my build output for comparison:

<details><summary>flutter build linux -v output</summary>

```
[   +3 ms] Building Linux application...
[   +9 ms] executing: [build/linux/x64/release/] cmake -G Ninja -DCMAKE_BUILD_TYPE=Release -DFLUTTER_TARGET_PLATFORM=linux-x64 /root/flutter_serial_port/example/linux
[ +186 ms] -- The CXX compiler identification is Clang 10.0.0
[   +8 ms] -- Check for working CXX compiler: /usr/bin/clang++
[ +141 ms] -- Check for working CXX compiler: /usr/bin/clang++ -- works
[   +1 ms] -- Detecting CXX compiler ABI info
[ +140 ms] -- Detecting CXX compiler ABI info - done
[  +13 ms] -- Detecting CXX compile features
[        ] -- Detecting CXX compile features - done
[   +5 ms] -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
[        ] -- Checking for module 'gtk+-3.0'
[  +22 ms] --   Found gtk+-3.0, version 3.24.20
[  +92 ms] -- Checking for module 'glib-2.0'
[  +16 ms] --   Found glib-2.0, version 2.64.6
[  +45 ms] -- Checking for module 'gio-2.0'
[  +20 ms] --   Found gio-2.0, version 2.64.6
[  +59 ms] -- Checking for module 'blkid'
[  +17 ms] --   Found blkid, version 2.34.0
[ +129 ms] -- The C compiler identification is Clang 10.0.0
[   +5 ms] -- Check for working C compiler: /usr/bin/clang
[  +85 ms] -- Check for working C compiler: /usr/bin/clang -- works
[   +1 ms] -- Detecting C compiler ABI info
[  +91 ms] -- Detecting C compiler ABI info - done
[  +12 ms] -- Detecting C compile features
[        ] -- Detecting C compile features - done
[   +3 ms] -- Configuring done
[  +14 ms] -- Generating done
[   +1 ms] -- Build files have been written to: /root/flutter_serial_port/example/build/linux/x64/release
[   +5 ms] executing: ninja -C build/linux/x64/release install
[  +11 ms] ninja: Entering directory `build/linux/x64/release'
[  +73 ms] [1/13] Building C object plugins/flutter_serial_port/libserialport/CMakeFiles/serialport.dir/__/__/third_party/libserialport/linux_termios.c.o
[  +49 ms] [2/13] Building C object plugins/flutter_serial_port/libserialport/CMakeFiles/serialport.dir/__/__/third_party/libserialport/timing.c.o
[  +19 ms] [3/13] Building C object plugins/flutter_serial_port/libserialport/CMakeFiles/serialport.dir/__/__/third_party/libserialport/linux.c.o
[ +572 ms] [4/13] Building C object plugins/flutter_serial_port/libserialport/CMakeFiles/serialport.dir/__/__/third_party/libserialport/serialport.c.o
[  +33 ms] [5/13] Linking C shared library plugins/flutter_serial_port/libserialport/libserialport.so
[+19343 ms] [6/13] Generating /root/flutter_serial_port/example/linux/flutter/ephemeral/libflutter_linux_gtk.so,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_basic_message_channel.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_binary_codec.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_binary_messenger.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_dart_project.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_engine.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_json_message_codec.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_json_method_codec.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_message_codec.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_method_call.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_method_channel.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_method_codec.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_method_response.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_plugin_registrar.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_plugin_registry.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_standard_message_codec.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_standard_method_codec.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_string_codec.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_value.h,
/root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/fl_view.h, /root/flutter_serial_port/example/linux/flutter/ephemeral/flutter_linux/flutter_linux.h, _phony_
[   +1 ms] [  +61 ms] executing: uname -m
[        ] [  +44 ms] Exit code 0 from: uname -m
[        ] [        ] x86_64
[        ] [  +16 ms] executing: [/root/flutter/] git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] [   +7 ms] Exit code 0 from: git -c log.showSignature=false log -n 1 --pretty=format:%H
[        ] [        ] 5e4e530e934a3a4bd4fff818c3f04193e36a0f7d
[        ] [   +1 ms] executing: [/root/flutter/] git tag --points-at 5e4e530e934a3a4bd4fff818c3f04193e36a0f7d
[        ] [  +12 ms] Exit code 0 from: git tag --points-at 5e4e530e934a3a4bd4fff818c3f04193e36a0f7d
[        ] [   +1 ms] executing: [/root/flutter/] git describe --match *.*.* --long --tags 5e4e530e934a3a4bd4fff818c3f04193e36a0f7d
[        ] [  +27 ms] Exit code 0 from: git describe --match *.*.* --long --tags 5e4e530e934a3a4bd4fff818c3f04193e36a0f7d
[... cut ...]
[        ] -- Installing: /root/flutter_serial_port/example/build/linux/x64/release/bundle/data/flutter_assets/fonts/MaterialIcons-Regular.otf
[   +1 ms] -- Installing: /root/flutter_serial_port/example/build/linux/x64/release/bundle/data/flutter_assets/AssetManifest.json
[        ] -- Installing: /root/flutter_serial_port/example/build/linux/x64/release/bundle/lib/libapp.so
[   +6 ms] Building Linux application... (completed in 22.0s)
[        ] "flutter linux" took 67,722ms.
[ +254 ms] ensureAnalyticsSent: 251ms
[   +1 ms] Running shutdown hooks
[        ] Shutdown hooks complete
[        ] exiting with code 0
```
</details>

digitalkirkley, 2021-03-17:
Really appreciated thank you - spotted it was the build-essentials.  All working now.

closed 2021-03-18 by jpnurmi