jpnurmi/flutter_libserialport #12: SerialPort Errors on Mac

SBS-FPATE-RINALDI, 2021-03-26:
Mac Version: Catalina 10.15.7, Mac mini (2018)

Forewarning that I am new to Flutter...

I've copied the same code from the example folder to use. It works perfectly on Windows, but when I try to run it on Mac I get this error: 

[image]

[image]

Any help is much appreciated. Thanks!

lvsecoto, 2021-03-30:
I got the same issues. Failed to open port，and it return –2.

lvsecoto, 2021-04-01:
I have found the solution, set `serialPort.configure` after `serialPort.open`

jpnurmi, 2021-04-01:
@lvsecoto Thanks for spending the time to figure it out! Sounds like that an assertion could help to catch the problem.

SBS-FPATE-RINALDI, 2021-04-01:
The example code in this repo doesn't use `serialPort.open`. I tried putting `serialPort.configure` after `final port = SerialPort(address);` but still ran into the same issue. What's weird is refreshing the app made it sometimes work, but most of the time threw the same error. I'll keep playing around on my end though.

lvsecoto, 2021-04-02:
> The example code in this repo doesn't use `serialPort.open`. I tried putting `serialPort.configure` after `final port = SerialPort(address);` but still ran into the same issue. What's weird is [...]

Try this
```
final port = SerialPort(address);
port.open(...);
port.configure =...
```

LiveRock, 2021-06-02:
```
══╡ EXCEPTION CAUGHT BY GESTURE ╞═══════════════════════════════════════════════════════════════════
The following SerialPortError was thrown while handling a gesture:
Invalid argument, errno = 22

When the exception was thrown, this was the stack:
#0      Util.call (package:libserialport/src/util.dart:39:7)
#1      Util.toInt (package:libserialport/src/util.dart:77:16)
#2      _SerialPortImpl.busNumber (package:libserialport/src/port.dart:284:17)
#3      new SerialPortDataSource.<anonymous closure> (package:vt_meter_app/sections/meters/meters.dart:301:40)
#4      MappedListIterable.elementAt (dart:_internal/iterable.dart:412:31)
#5      ListIterator.moveNext (dart:_internal/iterable.dart:341:26)
#6      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:188:27)
#7      new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
#8      new List.of (dart:core-patch/array_patch.dart:50:28)
#9      ListIterable.toList (dart:_internal/iterable.dart:212:44)
#10     new SerialPortDataSource (package:vt_meter_app/sections/meters/meters.dart:303:10)
#11     masterPaneMeters.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:vt_meter_app/sections/meters/meters.dart:166:60)
#12     State.setState (package:flutter/src/widgets/framework.dart:1088:30)
#13     masterPaneMeters.<anonymous closure>.<anonymous closure> (package:vt_meter_app/sections/meters/meters.dart:161:53)
#14     _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:989:21)
#15     GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:182:24)
#16     TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:607:11)
#17     BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:296:5)
#18     BaseTapGestureRecognizer.acceptGesture (package:flutter/src/gestures/tap.dart:267:7)
#19     GestureArenaManager.sweep (package:flutter/src/gestures/arena.dart:157:27)
#20     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:443:20)
#21     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:419:22)
#22     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:287:11)
#23     GestureBinding._handlePointerEventImmediately (package:flutter/src/gestures/binding.dart:374:7)
#24     GestureBinding.handlePointerEvent (package:flutter/src/gestures/binding.dart:338:5)
#25     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:296:7)
#26     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:279:7)
#30     _invoke1 (dart:ui/hooks.dart:182:10)
#31     PlatformDispatcher._dispatchPointerDataPacket (dart:ui/platform_dispatcher.dart:282:7)
#32     _dispatchPointerDataPacket (dart:ui/hooks.dart:96:31)
(elided 3 frames from dart:async)

```

I am still getting this error. What I want to do is just tor refresh the serial port list with
availablePorts = SerialPort.availablePorts;

xuwanjin, 2021-07-08:
> Try this
>
> ```
> final port = SerialPort(address);
> port.open(...);
> port.configure =...
> ```
thanks a lot, it works for me. I try it on MacOS, 11.4

jiangchunyu, 2021-11-11:
SerialPortConfig? config=_serialPort?.config;
          config?.baudRate=hdevice.baudRate;//波特率
          config?.stopBits = 1;//停止位
          config?.bits=8;//数据位
          config?.parity=0;//校验位
          config?.parity=0;
          if(config!=null){
            LogUtil.d(" config ${config}", tag: TAG);
            _serialPort?.config=config;
          }
这个方式可以。不报错

ehagerty, 2021-11-15:
@jiangchunyu 
谢谢你。您能否分享您的代码，以便我看看我的代码有什么不同？