jpnurmi/flutter_libserialport #15: Capture the Weight value of a Precision Scale

victorgarcia10, 2021-04-15:
Hello my friend! Nice package!

My name is Jhon and I'm a flutter developer.

I'm working in a desktop project that I need to check the connection and capture the weight of a precison scale connected to computer by Serial USB.

Can your package help me? 

I'm seeing that the package can check the connection and show information about  the Serial USB Dispositive. But the weight? Can I capture that?

jpnurmi, 2021-04-16:
Hi, do you know if [libserialport](https://sigrok.org/wiki/Libserialport) supports such functionality?

lucafabbri, 2024-02-21:
Hi Joao,

once you have the socket open you can get the weight accordingly to your device specs. Usually there is some protocol to implement in order to ask the scale to return the current weight.

I close the issue as no development is necessary

closed 2024-02-21 by lucafabbri

datpt11, 2024-10-02:
@joaovictorgarcia10 any update?