jpnurmi/flutter_libserialport #16: desktop version is not working for ubuntu 16.04 LTS

learncodingforweb, 2021-04-27:
```
/home/sukhdeep/project/git/bt/ft/webdev/nodejs/multiplatform/ft/menuplator/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/flutter_libserialport_plugin.cc:40:3: error: use of undeclared identifier 'setenv'; did you mean 'g_setenv'?
```

```
name: learn
description: A new Flutter project.
publish_to: 'none' # Remove this line if you wish to publish to pub.dev
version: 1.0.0+1

environment:
  sdk: ">=2.7.0 <3.0.0"

dependencies:
  flutter:
    sdk: flutter

  cupertino_icons: ^1.0.2
  flutter_libserialport: ^0.2.0

dev_dependencies:
  flutter_test:
    sdk: flutter

flutter:
  uses-material-design: true

```

jpnurmi, 2021-04-27:
Hi, I have published 0.2.1 with the fix proposed in the error message. Hopefully that helps.

closed 2021-04-27 by jpnurmi

learncodingforweb, 2021-04-28:
problem is resolved in 0.2.1

commit 5bf69491 (2021-04-27): Fix (g_)setenv() call

Close: #16