jpnurmi/flutter_libserialport #17: Not able to install app using flutter_libserial on android

learncodingforweb, 2021-04-28:
```
* What went wrong:
A problem occurred configuring project ':flutter_libserialport'.
> NDK at Android/Sdk/ndk/21.1.6352462 did not have a source.properties file
```
Even through updated android studio to use latest version of NDK

jpnurmi, 2021-04-28:
Hi, do I understand correctly that a plain Flutter counter example or such works fine, but as soon as flutter_libserialport is added as a dependency, the build error occurs?

learncodingforweb, 2021-04-29:
Problem get resolved by uninstall and after that again installing sdk in android resolves the problem.
Able to install app in android, But getting 

**SerialPortError: Permission denied, error = 13**

Running on Android Version 8.1.0
how to resolve this issue in android?

jpnurmi, 2021-04-29:
I don't know what kind of device you have, but hopefully a rooted one: https://github.com/jpnurmi/libserialport.dart/issues/3

lucafabbri, 2024-02-21:
We put permission handling in todo. 
Activity is tracked in #76 
Closed

closed 2024-02-21 by lucafabbri