jpnurmi/flutter_libserialport #18: Process on using the library

lolcatsnin, 2021-05-03:
Is there a set process to installing this manually ? 
ie: x in pubspec, add import to x, move this file to x ect ect

jpnurmi, 2021-05-06:
I'm afraid I don't understand the question. I don't think it's possible to use the API of a Dart package without adding it to pubspec and importing it.

jpnurmi, 2021-05-06:
The respective pure Dart package aka [`libserialport`](https://pub.dev/packages/libserialport) allows you to build/deploy/load the libserialport C-library as you wish if that's what you're asking.

lolcatsnin, 2021-05-12:
Hey sorry for the late reply. il have a gander at it again. i had got everything working nicely through the example i modified but bringing it all to a new project is what had stumped me so il give it another look.

closed 2021-05-12 by lolcatsnin

jpnurmi, 2021-05-12:
Now that I read the question again, I think the answer would be that no additional steps are required besides the usual pubspec.yaml changes and importing the lib. Oh, and the API is documented in the [libserialport](https://pub.dev/documentation/libserialport/latest/libserialport/libserialport-library.html) package.