jpnurmi/flutter_libserialport #19: Can this package be used to communicate with an Arduino Uno from a Flutter Desktop application? Link to any similar sample codes would be really helpful.

MrPengueeen, 2021-05-21:
I tried using the sample code to try and communicate with an Arduino Uno that just simply prints some string to serial port periodically. I couldn't get it to work, the data I receive are all empty. I would really appreciate any help regarding this. Thank you.

LiveRock, 2021-06-01:
I am interested in the answer too. I am trying to use this for a Flutter desktop app with ESP32

jiangweihu, 2021-11-25:
Have you solved this yet? I want use this for a Flutter desktop app with STM32.

nightlan1015297, 2022-09-26:
Having same issue.
But open the serial port in Arduino IDE first then open the port with this package everything works fine.

Did anyone solve this problem?

iqfareez, 2022-12-22:
Is this is what you guys are trying to achieve?

The counter is from the Arduino.

https://user-images.githubusercontent.com/60868965/209126724-684a16fe-0de9-4ede-912f-c6032046ab08.mp4

Super simple Arduino code:
```cpp
int i = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(i);
  i++;
  delay(1000);
}
```

Flutter (Dart) code:
```dart
import 'package:flutter/foundation.dart';
import 'package:flutter/material.dart';
import 'package:flutter_libserialport/flutter_libserialport.dart';

class ScorePage extends StatefulWidget {
  const ScorePage({super.key});

  @override
  State<ScorePage> createState() => _ScorePageState();
}

class _ScorePageState extends State<ScorePage> {
  SerialPort port = SerialPort('COM9');
  late SerialPortReader reader;

  @override
  void initState() {
    super.initState();
    var res = port.openRead();

    if (!res) {
      print('Error opening port:${port.name}');
      // handle to close reconnet
    }
    var portConfig = SerialPortConfig()
      ..baudRate = 9600
      ..bits = 8
      ..stopBits = 1;
    port.config = portConfig;

    reader = SerialPortReader(port);
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      body: Center(
        child: StreamBuilder<Uint8List>(
          stream: reader.stream,
          builder: (context, snapshot) {
            if (snapshot.hasError) {
              return Text('Error: ${snapshot.error}');
            }
            if (snapshot.connectionState == ConnectionState.waiting) {
              return const Text('Awaiting result...');
            }

            var decodedString = String.fromCharCodes(snapshot.data!);
            return Column(
              mainAxisAlignment: MainAxisAlignment.center,
              children: <Widget>[
                const Text(
                  'Counter from Arduino:',
                ),
                Text(
                  decodedString,
                  style: Theme.of(context).textTheme.headline1,
                ),
              ],
            );
          },
        ),
      ),
    );
  }

  @override
  void dispose() {
    port.close();
    super.dispose();
  }
}

```

Sometimes you may need to disconnect and reconnect the USB cable.

Zacchaeus-Oluwole, 2024-05-13:
> Is this is what you guys are trying to achieve?
> 
> The counter is from the Arduino.
> 
>  2022-12-22.19-39-18.mp4 
> Super simple Arduino code:
> 
> ```c++
> int i = 0;
> 
> void setup() {
>   // put your setup code here, to run once:
>   Serial.begin(9600);
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
>   Serial.println(i);
>   i++;
>   delay(1000);
> }
> ```
> 
> Flutter (Dart) code:
> 
> ```dart
> import 'package:flutter/foundation.dart';
> import 'package:flutter/material.dart';
> import 'package:flutter_libserialport/flutter_libserialport.dart';
> 
> class ScorePage extends StatefulWidget {
>   const ScorePage({super.key});
> 
>   @override
>   State<ScorePage> createState() => _ScorePageState();
> }
> 
> class _ScorePageState extends State<ScorePage> {
>   SerialPort port = SerialPort('COM9');
>   late SerialPortReader reader;
> 
>   @override
>   void initState() {
>     super.initState();
>     var res = port.openRead();
> 
>     if (!res) {
>       print('Error opening port:${port.name}');
>       // handle to close reconnet
>     }
>     var portConfig = SerialPortConfig()
>       ..baudRate = 9600
>       ..bits = 8
>       ..stopBits = 1;
>     port.config = portConfig;
> 
>     reader = SerialPortReader(port);
>   }
> 
>   @override
>   Widget build(BuildContext context) {
>     return Scaffold(
>       body: Center(
>         child: StreamBuilder<Uint8List>(
>           stream: reader.stream,
>           builder: (context, snapshot) {
>             if (snapshot.hasError) {
>               return Text('Error: ${snapshot.error}');
>             }
>             if (snapshot.connectionState == ConnectionState.waiting) {
>               return const Text('Awaiting result...');
>             }
> 
>             var decodedString = String.fromCharCodes(snapshot.data!);
>             return Column(
>               mainAxisAlignment: MainAxisAlignment.center,
>               children: <Widget>[
>                 const Text(
>                   'Counter from Arduino:',
>                 ),
>                 Text(
>                   decodedString,
>                   style: Theme.of(context).textTheme.headline1,
>                 ),
>               ],
>             );
>           },
>         ),
>       ),
>     );
>   }
> 
>   @override
>   void dispose() {
>     port.close();
>     super.dispose();
>   }
> }
> ```
> 
> Sometimes you may need to disconnect and reconnect the USB cable.

No, i need to perform write and read. read is working but write isn't working