jpnurmi/flutter_libserialport #20: flutter_libserialport does not work on Archlinux anymore

Kozzi11, 2021-05-26:
I have been usinig it for few weeks with no problem, but today it end up with these errors:

	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:98:9: error: invalid application of 'sizeof' to an incomplete type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:106:15: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:106:27: error: use of undeclared identifier 'RTSXOFF'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:107:15: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'CTSXON'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:108:15: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:108:27: error: use of undeclared identifier 'DTRXOFF'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:109:15: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:109:27: error: use of undeclared identifier 'DSRXON'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:118:7: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:118:22: error: use of undeclared identifier 'RTSXOFF'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:118:32: error: use of undeclared identifier 'CTSXON'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:118:41: error: use of undeclared identifier 'DTRXOFF'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:118:51: error: use of undeclared identifier 'DSRXON'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:121:8: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:121:21: error: use of undeclared identifier 'RTSXOFF'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:123:8: error: incomplete definition of type 'struct termiox'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:123:21: error: use of undeclared identifier 'CTSXON'
	/home/kozak/Vyvoj/Eurosat/etrackermanagetool/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:125:8: error: incomplete definition of type 'struct termiox'
	fatal error: too many errors emitted, stopping now [-ferror-limit=]
	Exception: Build process failed

Kozzi11, 2021-05-26:
It is issue with linux kernel version >=5.12, because in 5.12 they have removed some old dead code
https://github.com/torvalds/linux/blob/v5.12/include/uapi/linux/termios.h
vs
https://github.com/torvalds/linux/blob/v5.11/include/uapi/linux/termios.h

jpnurmi, 2021-05-26:
I guess we could temporarily rip off the related code from the local copy in `third_party/libserialport` but reporting the issue upstream would be better: https://sigrok.org/bugzilla/describecomponents.cgi?product=libserialport (requires registration).

Kozzi11, 2021-05-26:
this would not helped. I have try it. It would compile but still not works. But if I comment
#define HAVE_STRUCT_TERMIOX
in
~/.pub/hosted/pub.dartlang.org/flutter_libserialport-0.2.1/linux/libserialport/config.h

it works ok

PR #35 (2021-11-14): Fix linux builds for newer kernels
Fixes #20

Support for TERMIOX was removed from the Linux kernel starting in 2020-12. Apparently (see [linux mailing list](https://www.spinics.net/lists/linux-serial/msg41926.html)), TERMIOX was never used by any linux driver, but libserialport relies on it for some advanced flow control options (see [upstream bug](https://sigrok.org/bugzilla/show_bug.cgi?id=1687)).

Upstream has decided to disable checking for TERMIOX on linux ([patch](https://sigrok.org/gitweb/?p=libserialport.git;a=commit;h=6f9b03e597ea7200eb616a4e410add3dd1690cb1)). The kernel change has  [... cut]
commit fd575537 (2021-11-14): Fix linux builds for newer kernels

Fixes #20

Support for TERMIOX was removed from the Linux kernel starting in 2020-12. Apparently (see linux mailing list), TERMIOX was never used by any linux driver, but libserialport relies on it for some advanced flow control options (see upstream bug). Upstream has however decided to disable checking for TERMIOX on linux. The kernel change has been backported to the 5.10 LTS branch, so it doesn't only affect the lastest kernel versions.

Upstream Bug: https://sigrok.org/bugzilla/show_bug.cgi?id=1687
Upstream Path: https://sigrok.org/gitweb/?p=libserialpo [... cut]
closed 2021-11-14 by jpnurmi