jpnurmi/flutter_libserialport #22: API documentation and how to Send String instead of bytes?

LiveRock, 2021-06-15:
Some questions:
1. Where are the API documentation? I see many in the issues section here writing commands and methods thart are not mentioned at all in this github

2. How do I send strings via the serial port instead of sending bytes?
3. 
Thanks and hope to hear from you soon.

jpnurmi, 2021-06-15:
> 1. Where are the API documentation? I see many in the issues section here writing commands and methods thart are not mentioned at all in this github

[pub.dev](https://pub.dev/packages/flutter_libserialport) -> [API Reference](https://pub.dev/documentation/flutter_libserialport/latest/).

> 2. How do I send strings via the serial port instead of sending bytes?

[String.codeUnits](https://api.dart.dev/stable/2.10.5/dart-core/String/codeUnits.html)

closed 2021-06-15 by jpnurmi

LiveRock, 2021-06-16:
pub.dev -> API Reference
I have been the above but the docs are minimal.
There is no tutorial on how to proceed further other than a listing of all serial ports.
It would be good if there are examples like configurations, reading of data and also some documentation on all the methods and data.

Thanks