jpnurmi/flutter_libserialport #26: SerialPortError (SerialPortError: Permission denied, errno = 13)

workoutand, 2021-07-29:
[image]

Hi everyone,

I'm trying to run the example of the flutter_serialport lib, but it doesn't work on the android system, only windows, can anyone help me?

jpnurmi, 2021-07-30:
Hi, perhaps the discussion at https://github.com/jpnurmi/libserialport.dart/issues/3 helps

workoutand, 2021-07-30:
Hellow, I saw this thread but I didn't understand where those directories came from

laychopy, 2021-10-16:
Someone made it work?

josuemoraisgh, 2022-02-05:
Hello, I have the same problem and so far I haven't seen a solution on the forums. Can you help me?

ampcontrolgroup, 2022-03-20:
Anyone with a solution for this?

jeremylcarter, 2022-03-20:
Anyone with a solution for this?

Fudal, 2022-04-28:
I have the same problem

augustbering, 2022-05-02:
This lib should be updated to use the new android permissions handling, see https://github.com/altera2015/usbserial/blob/master/android/src/main/java/dev/bessems/usbserial/UsbSerialPlugin.java

That lib, usbserial, could be used instead of this one until it's fixed here I think.

trapeye, 2022-11-17:
I have the same problem, is it fix yet?

asrafilll, 2023-01-12:
still same problem guys, is it fix yet? Thanks

eldhosekm, 2023-07-07:
am also having the same problem. any fix?

augustbering, 2023-07-15:
only my "fix" from above

eldhosekm, 2023-07-16:
I make it work using the native implementation of Flutter. in native use this in gradle io.github.xmaihh:serialport:2.1.1

Muaz-fmb, 2023-10-25:
> I make it work using the native implementation of Flutter. in native use this in gradle io.github.xmaihh:serialport:2.1.1

is it possible for you to put the steps on how you solved it, unfortunately, I have zero knowledge of the native structure.

eldhosekm, 2023-10-29:
you can use MethodChannel to communicate to native here is the sample. You will get the callback in flutter from java or kotlin. 

https://medium.com/@iiharish97ii/method-channel-in-flutter-bridging-native-code-and-flutter-with-two-way-communication-788d1e91c8c1

then you write the code in native (java or kotlin) using this library( io.github.xmaihh:serialport:2.1.1) to communicate with serial port.

Muaz-fmb, 2023-10-29:
> you can use MethodChannel to communicate to native here is the sample. You will get the callback in flutter from java or kotlin. [...]

Can you help me with setting it up? it won't be for free. 
reach me out on mouazsysy@gmail.com

guyluz11, 2023-12-25:
Having the same problem while running flutter on Linux

lucafabbri, 2024-02-21:
We put permission handling in todo. 
Activity is tracked in #76 
Closed

closed 2024-02-21 by lucafabbri