jpnurmi/flutter_libserialport #28: `Count must be in the range` on Windows

xclud, 2021-09-30:
I get the error `Count must be in the range` as soon as I write to/read from serial port. I am using `SerialPortReader`. I am using Windows 10/11.

Getting this error on the first interaction is a blocking issue for our app.

I'd appreciate you help.

jpnurmi, 2021-09-30:
Do you have any code snippet to show? Where does the error come from? Does it show any stacktrace?

xclud, 2021-09-30:
I cloned this repo and modified `example/main.dart` like this:

```
void initPorts() {
  setState(() => availablePorts = SerialPort.availablePorts);

  final serial = SerialPort('COM1');

  if (serial.openReadWrite()) {
    final reader = SerialPortReader(serial);
    reader.stream.listen((event) {
      print(event);
    });

    serial.write(Uint8List.fromList([1, 2, 3, 4]));
  }
}
```

Stacktrace is null. But i have attached a screenshot:

[image]

AriaShishegaran, 2021-10-02:
@jpnurmi 
Any comment/progress on this? we are facing the same issue here. would appreciate it if you help us navigate through the issue.

jpnurmi, 2021-10-02:
I'm not sure why the library returns an error without providing the last error. Nevertheless, a sanity check like this could help. Are you able to test it for me? I don't have a suitable environment...

```diff
diff --git a/lib/src/util.dart b/lib/src/util.dart
index 2ca44c5..b0e2ddf 100644
--- a/lib/src/util.dart
+++ b/lib/src/util.dart
@@ -44,7 +44,7 @@ class Util {
   static Uint8List read(int bytes, UtilFunc<ffi.Uint8> readFunc) {
     final ptr = ffi.calloc<ffi.Uint8>(bytes);
     final len = call(() => readFunc(ptr));
-    final res = Uint8List.fromList(ptr.asTypedList(len));
+    final res = Uint8List.fromList(ptr.asTypedList(len.clamp(0, len)));
     ffi.calloc.free(ptr);
     return res;
   }
```

xclud, 2021-10-02:
The problem is with `len` being `-2`. I tried debugging but it seems `-2` comes from `serialport.dll`.

`final len = call(() => readFunc(ptr));`

jpnurmi, 2021-10-02:
Yes, it indicates an error but the last error is not set so it slips through. There was a reason to add that check so I'm sceptic to blindly change it. However, clamping the -2 length to 0 length would give and empty list, to make it a no-op.

AriaShishegaran, 2021-10-03:
> Yes, it indicates an error but the last error is not set so it slips through. There was a reason to add that check so I'm sceptic to blindly change it. However, clamping the -2 length to 0 length would give and empty list, to make it a no-op.

So do you consider this a bug that needs resolving? I'm asking this because initially, we viewed this as our own bad implementation, can we provide any extra data that helps you move forward with this bug or offer a workaround?

ricardoboss, 2022-12-01:
@jpnurmi @AriaShishegaran any progress on this one? I started getting this error after switching to `SerialPortReader`. Before, I was calling `port.read(...)` myself and had no such issue. The advantage of the reader is that it provides a convenient stream.

lucafabbri, 2024-02-21:
@ricardoboss we are about to release on pub dev the 0.4.0. 
Could check the issue against this release as available?

ricardoboss, 2024-02-21:
Sure, I can't say when I have time to test this though

lucafabbri, 2024-02-23:
> Sure, I can't say when I have time to test this though

No problem ;)