jpnurmi/flutter_libserialport #29: can not set baudrate

XAndroider, 2021-10-03:
when I set baudrate it does not work at all.

katyo, 2021-11-22:
I have same issue when I attempt to set `SerialPortConfig` on Linux:

```dart
port.config.baudRate = 9600;
```

```
No such file or directory, errno = 2

When the exception was thrown, this was the stack:
#0      Util.call (package:libserialport/src/util.dart:39:7)
#1      _SerialPortImpl.config (package:libserialport/src/port.dart:335:12)
...
```

```dart
var config = SerialPortConfig();
config.baudRate = 9600;
port.config = config;
```

```
No such file or directory, errno = 2

When the exception was thrown, this was the stack:
#0      Util.call (package:libserialport/src/util.dart:39:7)
#1      _SerialPortImpl.config= (package:libserialport/src/port.dart:347:10)
...
```

mortenboye, 2022-06-17:
Thanks @katyo for sharing that discovery. Setting the config after opening the port, solve my issue also. Great!

usernema01, 2023-07-24:
> 当我尝试`SerialPortConfig`在 Linux 上设置时，我遇到了同样的问题：
> 
> ```dart
> port.config.baudRate = 9600;
> ```
> 
> ```
> No such file or directory, errno = 2
> 
> When the exception was thrown, this was the stack:
> #0      Util.call (package:libserialport/src/util.dart:39:7)
> #1      _SerialPortImpl.config (package:libserialport/src/port.dart:335:12)
> ...
> ```
> 
> ```dart
> var config = SerialPortConfig();
> config.baudRate = 9600;
> port.config = config;
> ```
> 
> ```
> No such file or directory, errno = 2
> 
> When the exception was thrown, this was the stack:
> #0      Util.call (package:libserialport/src/util.dart:39:7)
> #1      _SerialPortImpl.config= (package:libserialport/src/port.dart:347:10)
> ...
> ```
> 
> UPD：正如我在`libserialport`示例中看到的，似乎应该在打开端口后设置配置。

thank you baby! I solve it! 太屌了

VinayakaKS, 2023-09-05:
Hi, I may be late to answer to this, but thinking it will help others who end up at this problem.
The port.config needs to be set only after opening the port.

      SerialPort port = SerialPort(availablePorts[widget.comPort]);

      final portconfig = SerialPortConfig();
      portconfig.baudRate = 115200;
      
      SerialPortReader reader = SerialPortReader(port, timeout: 10000);
     
      port.openReadWrite();
      port.config = portconfig;

lucafabbri, 2024-02-21:
Thanks @VinayakaKS I'll check it further to avoid issues when config is set before opening

dropik, 2024-08-08:
That's weird, but it indeed works if setting config takes place after .openReadWrite(), not before.

M0JXD, 2025-08-17:
I've spent an hour trying to debug this issue (and many days wondering why the port behaviour has been so inconsistent) just to find out the the config must be set after opening the port (which is backwards to what would be expected).

Could opening the port please be added to the example and possibly documented clearly somewhere? It seems many users have been troubled by this, and it's the root cause of many open issues both in this repo and the underlying [FFI repo](https://github.com/jpnurmi/libserialport.dart), e.g. [this one](https://github.com/jpnurmi/libserialport.dart/issues/24).

[The documentation stresses the importance of setting the config](https://pub.dev/documentation/flutter_libserialport/latest/flutter_libserialport/SerialPortConfig-class.html) but the example in FFI repo README doesn't at all.