jpnurmi/flutter_libserialport #30: SerialPortError on macOS: Operation not permitted, errno = 1 when trying to open the port

mkulesh, 2021-10-05:
HI, I have some trouble when I try to open serial port and write something into it on macOS. I use flutter_libserialport version 0.2.2.

My code is really simple:
```
final SerialPort sp = SerialPort("/dev/cu.usbserial-1420");
try
{
    if (sp.openWrite())
    {
        sp.config.baudRate = 115200;
        sp.config.parity = 0;
        sp.config.bits = 8;
        sp.config.stopBits = 1;
        sp.config.rts = 0;
        sp.config.dtr = 0;
        sp.write(Uint8List.fromList(rc.hex.codeUnits), timeout: 100);
        sp.close();
    }
    else
    {
        Logging.info(this, "error opening serial port: " + SerialPort.lastError.toString());
    }
}
on Exception catch (e)
{
    Logging.info(this, "can not write into serial port: " + SerialPort.lastError.toString());
}
sp.dispose();
```

However, I have an error when openWrite method is called:
```
flutter: onpc: #0201 RiCommandsMap: sending NUMBER_3 to USB-RI interface /dev/cu.usbserial-1420
flutter: onpc: #0202 RiCommandsMap: error opening serial port: SerialPortError: Operation not permitted, errno = 1
```

There are two ports available in the system:
```
family@mac-mini ~ % ls -la /dev/cu*                                                                       
crw-rw-rw-  1 root  wheel   19,   1  5 окт 20:47 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   19,   3  5 окт 20:47 /dev/cu.usbserial-1420
family@mac-mini ~ % whoami          
family
family@mac-mini ~ % id -Gn family   
staff everyone localaccounts _appserverusr admin _appserveradm _lpadmin com.apple.access_ssh com.apple.sharepoint.group.5 com.apple.sharepoint.group.1 com.apple.sharepoint.group.2 com.apple.sharepoint.group.4 _appstore _lpoperator _developer _analyticsusers com.apple.access_ftp com.apple.access_screensharing com.apple.access_remote_ae com.apple.sharepoint.group.3
```

My Flutter version is:
[✓] Flutter (Channel stable, 2.2.1, on macOS 11.0.1 20B29 darwin-x64, locale en-GB)
    • Flutter version 2.2.1 at /Volumes/ExtWork/home/family/work/android/flutter
    • Framework revision 02c026b03c (4 месяца назад), 2021-05-27 12:24:44 -0700
    • Engine revision 0fdb562ac8
    • Dart version 2.13.1

When I use, for example, this port in the "SerialTool" app with the same settings, it works as desired:
[image]

Could you please explain me that is the reason for such an error?

kadishmal, 2022-06-11:
@mkulesh your solution fixed perfectly. For newcomers the files are:

- `./macos/Runner/DebugProfile.entitlements`
- `./macos/Runner/Release.entitlements`

lucafabbri, 2024-02-21:
@kadishmal PRs are welcome ;)