jpnurmi/flutter_libserialport #34: sync error

GuiWolff, 2021-11-12:
Hello Friend!
I'm having trouble configuring the serial port.
When I test connecting an arduino through Putty, the Flutter serial connection works.
If I disconnect the arduino from the USB port and connect right after, the connection reads the data without sync.

My code:

  void conectarNaSerial(int address) {
      final port = SerialPort.fromAddress(address);
    port.open(mode: 1);
    if(port.isOpen) {
      port.config.baudRate = 115200;
      port.config.stopBits = 1;
      port.config.bits = 8;
      port.config.parity = SerialPortParity.none;
      port.config.setFlowControl(SerialPortFlowControl.none);

      reader = SerialPortReader(port);
      if(reader != null){
        reader!.stream.listen((d){
          print("${utf8.decode(d)}");
          var number = double.tryParse(utf8.decode(d));
          if(number == null){
            return;
          } else {
            valorSerial.value = number;
          }
        });
      }
    }
  }

I must configure the serial port right after connecting, right?

moodstubos, 2022-11-30:
@GuiWolff
Your post was long time ago, but I have exactly the same issue. 
After arduino connect to USB I have to connect serial port with Putty first. After that I am able to receive data via flutter. If not using putty first, this package does not receive any data at all. 

did you find any solution for this?

lucafabbri, 2024-02-21:
@moodstubos @GuiWolff I guess #29 might help you