jpnurmi/flutter_libserialport #36: Failed to load dynamic library 'libserialport.dylib'

YowFung, 2021-12-01:
My Code:
```dart
import 'package:flutter_libserialport/flutter_libserialport.dart';

void main() {
  print(SerialPort.availablePorts);
}
```

Error Log:
```log
Testing started at 15:53 ...
dart:ffi                                    new DynamicLibrary.open
package:libserialport/src/dylib.dart 32:54  dylib
package:libserialport/src/port.dart 230:32  _SerialPortImpl.availablePorts.<fn>
package:libserialport/src/util.dart 37:21   Util.call
package:libserialport/src/port.dart 230:21  _SerialPortImpl.availablePorts
package:libserialport/src/port.dart 82:61   SerialPort.availablePorts
test/exploration/serial.dart 4:20           main

Failed to load "/Users/yowfung/workspace/proitav/projects/ApiTestTool/src/common/test/exploration/serial.dart": Invalid argument(s): Failed to load dynamic library 'libserialport.dylib': dlopen(libserialport.dylib, 0x0001): tried: '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/./libserialport.dylib' (no such file), '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/../../../libserialport.dylib' (no such file), '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/Frameworks/libserialport.dylib' (no such file), '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/./libserialport.dylib' (no such file), '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/../../../libserialport.dylib' (no such file), '/Users/yowfung/Library/Flutter/bin/cache/dart-sdk/bin/Frameworks/libserialport.dylib' (no such file), 'libserialport.dylib' (no such file), '/usr/lib/libserialport.dylib' (no such file), '/Users/yowfung/workspace/proitav/projects/ApiTestTool/src/common/libserialport.dylib' (no such file), '/usr/lib/libserialport.dylib' (no such file)

Process finished with exit code 1

```

Description of this plug-in on [pub.dev](https://pub.flutter-io.cn/packages/flutter_libserialport#serial-port-for-flutter):

> `flutter_libserialport` is a simple wrapper around the `libserialport` Dart package, utilizing Flutter's build system to build and deploy the `libserialport` C-library under the hood. This package does not provide any additional API, but merely helps to make the `libserialport` Dart package work "out of the box" without the need of manually building and deploying the `libserialport` C-library.

Do I need any additional configuration?

<details>
<summary>flutter doctor -v</summary>

```log
[✓] Flutter (Channel stable, 2.5.3, on macOS 12.0.1 21A559 darwin-x64, locale zh-Hans-CN)
    • Flutter version 2.5.3 at /Users/yowfung/Library/Flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 18116933e7 (7 weeks ago), 2021-10-15 10:46:35 -0700
    • Engine revision d3ea636dc5
    • Dart version 2.14.4
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[✓] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at /Users/yowfung/Library/Android/sdk
    • Platform android-31, build-tools 31.0.0
    • ANDROID_HOME = /Users/yowfung/Library/Android/sdk
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 13.1, Build version 13A1030d
    • CocoaPods version 1.10.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)

[✓] IntelliJ IDEA Community Edition (version 2021.2.2)
    • IntelliJ at /Applications/IntelliJ IDEA CE.app
    • Flutter plugin version 61.2.4
    • Dart plugin version 212.5486

[✓] VS Code (version 1.57.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.25.0

[✓] Proxy Configuration
    • HTTP_PROXY is set
    • NO_PROXY is
      localhost,0.0.0.0,127.0.0.1,192.168.1.253,192.168.1.250,192.168.1.251,192.168.5.253,192.168.0.
      122,192.168.*,169.254.1.1,169.254.*
    • NO_PROXY contains 127.0.0.1
    • NO_PROXY contains localhost

[✓] Connected device (2 available)
    • macOS (desktop) • macos  • darwin-x64     • macOS 12.0.1 21A559 darwin-x64
    • Chrome (web)    • chrome • web-javascript • Google Chrome 96.0.4664.55

• No issues found!
```
</details>

Zopenzop, 2022-04-17:
@YowFung I am facing a similar problem, did you ever get a solution to this?

YowFung, 2022-04-21:
@Zopenzop 
已经找到解决办法了，请在 XCode 中勾选 USB 设备访问权限。

closed 2022-04-21 by YowFung

Zopenzop, 2022-04-22:
Can you share the code from your android manifest file so I can crosscheck whether mine's correct?

YowFung, 2022-04-22:
@Zopenzop 
我没有在安卓上试过，我开发的是桌面应用程序，所以只在 Windows 和 macOS 平台上使用这个插件。
安卓的话，我想应该也是需要在 Manifest 文件中加入访问 USB 设备的权限吧？

Zopenzop, 2022-04-22:
Ahh I see, I am actually trying to use flutter_libserialport on Android 11. Haven't been able to devise a solution to this issue yet which doesn't seem to be an issue with permissions, might be something else. I also tried a few other serial communication plugins, which didn't work either due to some other issues I encountered. The support for android on the internet is still pretty shallow as far as my experience goes, seems like I'll have to get my hands dirty and diagnose it all. Anyways, thanks for replying.

Zopenzop, 2022-04-22:
@YowFung Diagnosed, I was just missing Visual Studio and he C++ development dependencies.