jpnurmi/flutter_libserialport #42: [M1 macOS 12.1] Calling Serialport.dispose() crashes app

smofe, 2022-01-25:
When I try to call the `dispose()` method of a `SerialPort` object, the app crashes. MacOS generates the following error report:

```
OS Version:            macOS 12.1 (21C52)
...
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
<appName>(2663,0x304647000) malloc: *** error for object 0x600002dee4c0: pointer being freed was not allocated
abort() called
```

This happens when I execute the following method:
``` dart
Future<void> disconnect() async {
    try {
      await sendString(data: "cliStop");
    } catch (e) {
      print(e);
    }

    _reader?.close();
    port.close();
    port.dispose(); // this is the line which crashes the app.

    super.disconnect();
  }
```

Flutter doctor:
<details>

  <summary>Click to show</summary>

```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.8.1, on macOS 12.1 21C52 darwin-arm, locale de)
[✓] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] Connected device (2 available)
```
</details>

Do you have any idea why this is happening? Any help is very much appreciated!

smofe, 2022-01-28:
I've noticed that this only happens when I dispose the SerialPortConfig before. What I do when I connect to the device:
``` dart
port.openReadWrite();

SerialPortConfig config = SerialPortConfig();
config.baudRate = baudRate;
config.bits = 8;
config.stopBits = 1;
config.parity = SerialPortParity.none;
config.setFlowControl(SerialPortFlowControl.xonXoff);

port.config = config;
print("Baud rate: ${port.config.baudRate}, "
      "Bits: ${port.config.bits}, "
      "Stop bits: ${port.config.stopBits}, "
      "Parity: ${port.config.parity}, "
      "XonXoff: ${port.config.xonXoff}");
config.dispose();
```

Removing the `config.dispose()` fixes the problem with the app crashing when calling the disconnect method described above. Can someone explain to me why this is happening and if this is expected behaviour?

printfinn, 2023-11-14:
Did you want `port.dispose()` instead of `config.dispose()`?

lucafabbri, 2024-02-21:
Closed as disposing a closed serial throws exception is an expected behavior from libserialport

closed 2024-02-21 by lucafabbri