jpnurmi/flutter_libserialport #44: [Windows] Cannot build my application when using this plugin

YowFung, 2022-02-17:
`flutter run -d windows`:

```txt
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Launching lib\main.dart on Windows in debug mode...
Building Windows application...
flutter_libserialport_plugin.obj : error LNK2019: 无法解析的外部符号 __imp__invalid_parameter，函数 "void * __cdecl std::_Allocate_manually_vector_aligned<struct std::_Default_allocate_traits>(unsigned __int64)" (??$_Allocate_manually_vector_aligned@U_Default_allocate_traits@std@@@std@@YAPEAX_K@Z) 中引用了该符号 [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(plugin_registrar.obj) : error LNK2001: 无法解析的外部符号 __imp__invalid_parameter [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(core_implementations.obj) : error LNK2001: 无法解析的外部符号 __imp__invalid_parameter [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_libserialport_plugin.obj : error LNK2019: 无法解析的外部符号 __imp__CrtDbgReport，函数 "void * __cdecl std::_Allocate_manually_vector_aligned<struct std::_Default_allocate_traits>(unsigned __int64)" (??$_Allocate_manually_vector_aligned@U_Default_allocate_traits@std@@@std@@YAPEAX_K@Z) 中引用了该符号 [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(plugin_registrar.obj) : error LNK2001: 无法解析的外部符号 __imp__CrtDbgReport [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(core_implementations.obj) : error LNK2001: 无法解析的外部符号 __imp__CrtDbgReport [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\Debug\flutter_libserialport_plugin.dll : fatal error LNK1120: 2 个无法解析的外部命令 [D:\workspace\proitav\git\APITester\src\gui\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
Exception: Build process failed.
```

`flutter --version`:
```txt
Flutter 2.10.1 • channel stable • https://github.com/flutter/flutter.git
Framework • revision db747aa133 (8 days ago) • 2022-02-09 13:57:35 -0600
Engine • revision ab46186b24
Tools • Dart 2.16.1 • DevTools 2.9.2
```

ciborg971, 2022-11-29:
same error on Windows 11 Arm64 :

`flutter run -d windows
Running "flutter pub get" in printconf...                           4.9s
Launching lib\main.dart on Windows in debug mode...
flutter_libserialport_plugin.obj : error LNK2019: unresolved external symbol __imp_FlutterDesktopPluginRegistrarSetDestructionHandler referenced in function "public: class flutter::PluginRegistrarWindows * __cdecl flutter::PluginRegistrarManager::GetRegistrar<class flutter::PluginRegistrarWindows>(struct FlutterDesktopPluginRegistrar *)" (??$GetRegistrar@VPluginRegistrarWindows@flutter@@@PluginRegistrarManager@flutter@@QEAAPEAVPluginRegistrarWindows@1@PEAUFlutterDesktopPluginRegistrar@@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_libserialport_plugin.obj : error LNK2019: unresolved external symbol __imp_FlutterDesktopPluginRegistrarGetView referenced in function "public: __cdecl flutter::PluginRegistrarWindows::PluginRegistrarWindows(struct FlutterDesktopPluginRegistrar *)" (??0PluginRegistrarWindows@flutter@@QEAA@PEAUFlutterDesktopPluginRegistrar@@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(plugin_registrar.obj) : error LNK2019: unresolved external symbol __imp_FlutterDesktopPluginRegistrarGetMessenger referenced in function "public: __cdecl flutter::PluginRegistrar::PluginRegistrar(struct FlutterDesktopPluginRegistrar *)" (??0PluginRegistrar@flutter@@QEAA@PEAUFlutterDesktopPluginRegistrar@@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(plugin_registrar.obj) : error LNK2019: unresolved external symbol __imp_FlutterDesktopRegistrarGetTextureRegistrar referenced in function "public: __cdecl flutter::PluginRegistrar::PluginRegistrar(struct FlutterDesktopPluginRegistrar *)" (??0PluginRegistrar@flutter@@QEAA@PEAUFlutterDesktopPluginRegistrar@@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(core_implementations.obj) : error LNK2019: unresolved external symbol __imp_FlutterDesktopMessengerSend referenced in function "public: virtual void __cdecl flutter::BinaryMessengerImpl::Send(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned char const *,unsigned __int64,class std::function<void __cdecl(unsigned char const *,unsigned __int64)>)const " (?Send@BinaryMessengerImpl@flutter@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEBE_KV?$function@$$A6AXPEBE_K@Z@4@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(core_implementations.obj) : error LNK2019: unresolved external symbol __imp_FlutterDesktopMessengerSendWithReply referenced in function "public: virtual void __cdecl flutter::BinaryMessengerImpl::Send(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned char const *,unsigned __int64,class std::function<void __cdecl(unsigned char const *,unsigned __int64)>)const " (?Send@BinaryMessengerImpl@flutter@@UEBAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEBE_KV?$function@$$A6AXPEBE_K@Z@4@@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
flutter_wrapper_plugin.lib(core_implementations.obj) : error LNK2019: unresolved external symbol __imp_FlutterDesktopMessengerSendResponse referenextureFrameAvailable referenced in function "public: virtual bool __cdecl flutter::TextureRegistrarImpl::MarkTextureFrameAvailable(__int64)" (?MarkTextureFrameAvailable@TextureRegistrarImpl@flutter@@UEAA_N_J@Z) [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
C:\Users\crylog\project\Malt_IMAC\printconf\windows\flutter\ephemeral\flutter_windows.dll.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'ARM64' [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\Debug\flutter_libserialport_plugin.dll : fatal error LNK1120: 11 unresolved externals [C:\Users\crylog\project\Malt_IMAC\printconf\build\windows\plugins\flutter_libserialport\flutter_libserialport_plugin.vcxproj]
Building Windows application...Exception: Build process failed.`

`
flutter --version
Flutter 3.3.9 • channel stable • https://github.com/flutter/flutter.git
Framework • revision b8f7f1f986 (7 days ago) • 2022-11-23 06:43:51 +0900
Engine • revision 8f2221fbef
Tools • Dart 2.18.5 • DevTools 2.15.0
`
and Windows 11 x86-64 (error is in french but it's exactly the same)

lucafabbri, 2024-02-21:
Fixed in 0.4.0

closed 2024-02-21 by lucafabbri