jpnurmi/flutter_libserialport #49: Serial port hot plug

ZKXU, 2022-03-26:
Hi, does this lib support the serial port hot plug? if not, could you please help to give me some suggestion about how to implement this? Thanks for your support!

lucafabbri, 2024-02-21:
Could you provide more info about what your are trying to achieve?

maks, 2024-03-15:
I think what is being asked for is to have a stream to listen to for changes in state for when a device is attached too or disconnected from a serial port.

JabalnurIT, 2024-03-19:
yes, im troubled with this too

ElHyperion, 2024-04-29:
I would also be interested in this functionality, so that I wouldn't have to periodically check all opened ports before I connect my device.

maks, 2024-05-01:
Its Linux specific, but I was able to add basic support for my application using the udev package from Canonical to listen for USB device hotplug, though of course this won't work with non Linux OS's or non USB UARTs, but at least for my case it works.