jpnurmi/flutter_libserialport #51: Baud Rate Configuration fails silently

benjaminpaik, 2022-05-05:
I'm currently porting an app from Java/Swing to Flutter that running on Windows 11 that communicates with devices via serial. For the most part, everything works as expected. Occasionally, however, the baud rate doesn't set correctly. No errors are reported and calling the baud rate getter after opening the port returns the intended baud rate. Measuring the baud rate with an oscilloscope shows 9600 regardless of what I actually set the baud rate to in the code. The problem persists even after closing and reopening the app, restarting my PC, swapping USB/Serial converters, and changing which USB port the converter is plugged into. Interestingly, I can resolve the issue by launching the old Java/Swing app and opening the COM port. Once I have done this, the Flutter app again works as expected (for a couple of days) before the issue arises again. I've posted a snippet of my initialization code below for reference.

`        
try {
  comPort = SerialPort(serialPort);
  comPort.config = SerialPortConfig()
       ..baudRate = baudRate
       ..parity = parity
       ..bits = bits
       ..stopBits = stopBits;
  comPort.openReadWrite();
 } catch (e) {
     print(e.toString());
}
`

mortenboye, 2022-06-17:
See #29 for a possible solution:
`UPD: As I seen in libserialport examples, seems the config should be set after opening the port.`

benjaminpaik, 2022-06-20:
Thanks. It looks like your suggestion worked for me.

closed 2022-06-20 by benjaminpaik