jpnurmi/flutter_libserialport #53: Length must be in range

Zopenzop, 2022-06-03:
My code is as follows, pretty basic

```
        availablePorts = libserial.SerialPort.availablePorts;
        libserial.SerialPort port = libserial.SerialPort(availablePorts.first);
        port.openWrite();

        var config = port.config;
        config.baudRate = 9600;
        port.config = config;

        port.config.bits = 8;
        port.config.stopBits = 1;
        port.config.parity = 0;

        port.config = config;
        final reader = libserial.SerialPortReader(port);
        reader.stream.listen((Uint8List data) {
          log(utf8.decode(data));
        });
```

I get an error on receiving data -

```
Invalid argument(s): length must be in the range [0, 4611686018427387903].
#0      _checkExternalTypedDataLength (dart:ffi-patch/ffi_patch.dart:60:5)
#1      Uint8Pointer.asTypedList (dart:ffi-patch/ffi_patch.dart:620:5)
#2      Util.read
package:libserialport/src/util.dart:47
#3      _SerialPortReaderImpl._waitRead
package:libserialport/src/reader.dart:145
#4      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:300:17)
#5      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:192:12)
```

omidFdisk, 2022-07-22:
i have same issue

aboutZZ, 2022-09-24:
Fixed. The serial port device driver is not compatible with Windows 10, please update the driver.
Go to Settings -> privacy -> other devices, check your device is on the trusted device list.
[image]
OR use Windows 7😄

Zopenzop, 2022-10-12:
Apparently, the reason why I was not able to get data correctly / was getting zero nyte arrays was because I was setting the port config **after**  opening the serial port. What solved the issue was setting the port config **before** opening the serial port.

closed 2022-10-12 by Zopenzop

Ankur261, 2023-01-12:
> Apparently, the reason why I was not able to get data correctly / was getting zero byte arrays was because I was setting the port config **after** opening the serial port. What solved the issue was setting the port config **before** opening the serial port.

Can you please show the code or can tell me where to put the configurations.

Ankur261, 2023-01-12:
> aboutZZ
USB option is not showing on my system. Is there anything I need to change ?