jpnurmi/flutter_libserialport #54: linux error no  22 when config with baudrate

toplinuxsir, 2022-06-29:
```
 final portReader = SerialPortReader("/dev/ttyACM0");

  try {
    port.config.baudRate = baudrate;
    port.config.bits = dataBits;
    port.config.parity = parity;
    port.config.stopBits = stopBits;
  } catch (e, s) {
    debugPrint("error : $e");

    debugPrint("stack trace: $s");
  }
```

error as below 
```
 SerialPortError: Invalid argument, errno = 22
flutter: stack trace: #0      Util.call (package:libserialport/src/util.dart:39:7)
flutter: #1      _SerialPortImpl.config (package:libserialport/src/port.dart:335:12)
```

toplinuxsir, 2022-06-29:
close the issue !
I got it 
The port should open before configure .

lucafabbri, 2024-02-21:
#29

closed 2024-02-21 by lucafabbri