jpnurmi/flutter_libserialport #55: There are ports that cannot be read.

sm9i, 2022-07-08:
os:mac

sm9i, 2022-07-08:
kotlin code and console:
```kotlin
fun getAllDevice() {
    val portList = CommPortIdentifier.getPortIdentifiers()
    while (portList.hasMoreElements()) {
        val device: CommPortIdentifier = portList.nextElement() as CommPortIdentifier
        println("可用端口:${device.name}")
    }
}
```
console:
```console
可用端口:/dev/tty.BLTH
可用端口:/dev/cu.BLTH
可用端口:/dev/tty.Bluetooth-Incoming-Port
可用端口:/dev/cu.Bluetooth-Incoming-Port
可用端口:/dev/tty.usbserial-14210
可用端口:/dev/cu.usbserial-14210
```

dart code:
```dart
  void initPorts() {
    setState(() => availablePorts = SerialPort.availablePorts);
    availablePorts.forEach((item) {
      print(item);
    });
  }
```
console:
```console
flutter: /dev/cu.BLTH
flutter: /dev/cu.Bluetooth-Incoming-Port
flutter: /dev/cu.usbserial-14210
```

lucafabbri, 2024-02-21:
@sm9i dev/tty represent the current terminal, I think the answer lays in the way CommPortIdentifier ask the os for the list of devices. In your case in kotlin you have 2 file descriptors for the same resource, while in Flutter only one. The flutter behavior seems to me more clear and safe.

Closed

closed 2024-02-21 by lucafabbri