jpnurmi/flutter_libserialport #59: Fails to build when using a different PUB_HOSTED_URL

LeoBound, 2022-08-24:
Hi,

I use Artifactory to create a cache of all the dependencies I use for building my flutter apps. When building android apps from the pub cached copy of flutter_libserialport the build fails.

I can get around this for the time being by changing my pubspec entry from 
```
flutter_libserialport: ^0.2.3
``` 
to
```  
flutter_libserialport: 
    hosted: https://pub.dartlang.org
    version: ^0.2.3
```

But this isn't ideal as the package won't be cached.

I haven't tried this with other pub caching/hosting tools, I wonder if it could be related to `-` characters in the URL being replaced with `%47`, but this is purely speculation.

The build log error is as follows:

<details>
  <summary>Build Output</summary>

```  
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:externalNativeBuildRelease'.
> Build command failed.
  Error while executing process /Users/lbound/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/.cxx/cmake/release/armeabi-v7a serialport}
  ninja: Entering directory `/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/.cxx/cmake/release/armeabi-v7a'
  [1/5] Building C object 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c.o'
  FAILED: CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c.o 
  /Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang --target=armv7-none-linux-androideabi16 --gcc-toolchain=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -Oz -DNDEBUG  -fPIC   -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c.o' -MF CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c.o.d -o 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c.o'   -c '/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux_termios.c'
  depfile has multiple output paths[2/5] Building C object 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c.o'

  FAILED: CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c.o 
  /Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang --target=armv7-none-linux-androideabi16 --gcc-toolchain=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -Oz -DNDEBUG  -fPIC   -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c.o' -MF CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c.o.d -o 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c.o'   -c '/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/timing.c'
  depfile has multiple output paths[3/5] Building C object 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c.o'

  FAILED: CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c.o 
  /Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang --target=armv7-none-linux-androideabi16 --gcc-toolchain=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/lbound/Library/Android/sdk/ndk/21.1.6352462/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport -I/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -Oz -DNDEBUG  -fPIC   -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c.o' -MF CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c.o.d -o 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c.o'   -c '/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_libserialport-0.2.3/third_party/libserialport/linux.c'
  depfile has multiple output paths[4/5] Building C object 'CMakeFiles/serialport.dir/Users/lbound/Documents/flutter/.pub-cache/hosted/your.pub-url.here47artifactory%47api%47pub%47repository%47/flutter_l [... cut ...] erialport.c'
  depfile has multiple output paths
  ninja: build stopped: subcommand failed.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 25s
Running Gradle task 'assembleRelease'...                           26.0s
Gradle task assembleRelease failed with exit code 1
```  

</details>

Let me know if you need any more details 🙂

lucafabbri, 2024-02-21:
Hi @LeoBound unfortunately I can't reproduce your issue as I'm used to artifactory.
Did you managed to turn around this problem?

LeoBound, 2024-02-22:
Hi @lucafabbri ,
For the time being I've been referencing the package on pub.dev as in my initial comment.
I've updated to 0.4.0 of this package and the same issue persists. I'll see if I can have a look into it but to be honest it's not been that high priority as the workaround is easy.

lucafabbri, 2024-02-23:
Ok thanks, I'll keep the issue open. Should you have any further info it is always welcome ;)