jpnurmi/flutter_libserialport #60: Android - Permission denied error

shashidhar34, 2022-09-20:
Throws permission denied error when calling SerialPort.availablePorts method on Android. I have tried adding the intent filer and meta data fields as per [Android USB Host](https://developer.android.com/guide/topics/connectivity/usb/host#manifest-example) but it still throws the same error.

E/flutter ( 8439): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: SerialPortError: Permission denied, errno = 13
E/flutter ( 8439): #0      Util.call (package:libserialport/src/util.dart:40:7)
E/flutter ( 8439): #1      _SerialPortImpl.availablePorts (package:libserialport/src/port.dart:230:21)
E/flutter ( 8439): #2      SerialPort.availablePorts (package:libserialport/src/port.dart:82:61)
E/flutter ( 8439): #3      _ExampleAppState.initPorts.<anonymous closure> (package:handheldmultiplatform/main.dart:41:48)
E/flutter ( 8439): #4      State.setState (package:flutter/src/widgets/framework.dart:1114:30)
E/flutter ( 8439): #5      _ExampleAppState.initPorts (package:handheldmultiplatform/main.dart:41:5)
E/flutter ( 8439): #6      _InkResponseState.handleTap (package:flutter/src/material/ink_well.dart:1072:21)
E/flutter ( 8439): #7      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:253:24)
E/flutter ( 8439): #8      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:627:11)
E/flutter ( 8439): #9      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:306:5)
E/flutter ( 8439): #10     BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:239:7)
E/flutter ( 8439): #11     PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:615:9)
E/flutter ( 8439): #12     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:98:12)
E/flutter ( 8439): #13     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:143:9)
E/flutter ( 8439): #14     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:617:13)
E/flutter ( 8439): #15     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:141:18)
E/flutter ( 8439): #16     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:127:7)
E/flutter ( 8439): #17     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:460:19)
E/flutter ( 8439): #18     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:440:22)
E/flutter ( 8439): #19     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:337:11)
E/flutter ( 8439): #20     GestureBinding._handlePointerEventImmediately (package:flutter/src/gestures/binding.dart:395:7)
E/flutter ( 8439): #21     GestureBinding.handlePointerEvent (package:flutter/src/gestures/binding.dart:357:5)
E/flutter ( 8439): #22     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:314:7)
E/flutter ( 8439): #23     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:295:7)
E/flutter ( 8439): #24     _invoke1 (dart:ui/hooks.dart:167:13)
E/flutter ( 8439): #25     PlatformDispatcher._dispatchPointerDataPacket (dart:ui/platform_dispatcher.dart:341:7)
E/flutter ( 8439): #26     _dispatchPointerDataPacket (dart:ui/hooks.dart:94:31)

momo2555, 2022-09-26:
I had exactly the same error

Lostaroth, 2022-09-26:
Same error here.

trapeye, 2022-11-17:
same, I get same error

Wichnesh, 2022-12-23:
I'm also Facing this same error. 
If Anyone have solution for this error.pls replay..

MostafaDadkhah, 2022-12-24:
> I'm also Facing this same error. If Anyone have solution for this error.pls replay..

I had this problem in the emulator and I did not have such a problem in the real device.

andrebadini, 2023-01-07:
For cross-platform purposes, I recommend the "usb_serial" lib for Android, which works, while the "Libserialport" doesn't work.

magnus-lpa, 2023-06-22:
Have this problem as well, anyone have any updates?

eldhosekm, 2023-07-08:
i also have the same problem in arm64

magnus-lpa, 2023-07-10:
You could try using usb_serial plugin instead, I got that one working by using the latest commit from Github instead of the release on pub.dev. See my comment on a similar question: https://github.com/jpnurmi/flutter_libserialport/issues/76#issuecomment-1609026132

lucafabbri, 2024-02-21:
We put permission handling in todo. 
Activity is tracked in #76 
Closed

closed 2024-02-21 by lucafabbri