jpnurmi/flutter_libserialport #61: android

newer-android, 2022-10-18:
when i use it , tips
/dev/ttyS1
E/flutter ( 4259): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: SerialPortError: No such file or directory, errno = 2

newer-android, 2022-10-18:
my codes

 final name = SerialPort.availablePorts[1];

    final port = SerialPort(name);
    final serialPortConfig = SerialPortConfig();
    serialPortConfig.baudRate = 9600;
    port.config = serialPortConfig;
    if (!port.openReadWrite()) {
      print(SerialPort.lastError);
      exit(-1);
    }

    final reader = SerialPortReader(port);
    reader.stream.listen((data) {
      print('received: $data');
    });

lucafabbri, 2024-02-21:
@newer-android dev/tty is descriptor that refers to the terminal itself. I guess the is that the descriptor id deleted when you close the terminal. You should look for a descriptor with the same name nut without tty (as an example dev/cu, which is current user)

closed 2024-02-21 by lucafabbri