jpnurmi/flutter_libserialport #62: USB port not listening on connect - Windows

manishbuttan, 2022-10-29:
Hi, when I open the port in flutter to my USB serial port, it opens fine. But the port won't listen for incoming messages. If I close the port in the app and open it from Tera Term and close it, then open it again in the app, the communcation works fine and the incoming messages are working.

What is the proper way of opening the port and starting communication with the serial port without opening it externally?

Platform : Windows. Flutter.

if (!port.openReadWrite()) {
print("Port not opened: ${SerialPort.lastError}");
}

try {
serialPortReader = SerialPortReader(port);
serialPortReader.stream.listen((data) {
String message = String.fromCharCodes(data);
_processUSBData(message);
});
} catch (e) {
print(SerialPort.lastError);
}

Ibrahim653, 2022-11-10:
same issue here

masterix21, 2022-11-18:
Try applying the port config when your port is already opened. Like so:
```dart
var port = SerialPort(gateway.port);
port.openReadWrite();

port.config = SerialPortConfig()
      ..baudRate = 9600
      ..bits = 8
      ..stopBits = 1
      ..parity = SerialPortParity.none
      ..setFlowControl(SerialPortFlowControl.none);
```

moodstubos, 2022-11-30:
same here. 
I have to open serial port with putty first. After that I am able to receive data with this package. Without no data are received.
Testet on windows 11

manishbuttan, 2022-12-08:
Thank you moodstubos. These settings worked for me as well. Import information to add to the documentation.

sattha, 2023-01-04:
set the config after `openReadWrite` or other "open" port methods work for me.

NeariX67, 2023-06-19:
Sattha is right here. The SerialPortConfig class documentation states the following:
`@note A port must be opened before you can change its settings.`

Solved my issue as well :)

Ibrahim653, 2023-10-13:
hahahahahahahaha

closed 2024-02-21 by lucafabbri

Pedro-Sarmentoo, 2024-10-23:
I'm using Windows 11 and an ESP32 and I was having the same problem, flutter connected to the port but didn't receive data, and this solved my problem, my code worked like this:

        if (Varcontroller.instance.port!.openReadWrite()) {
        // Configurar a porta serial
        Varcontroller.instance.port?.config = SerialPortConfig()
        ..baudRate = 115200
        ..bits = 8
        ..stopBits = 1
        ..parity = SerialPortParity.none
        ..rts = SerialPortRts.flowControl
        ..cts = SerialPortCts.flowControl
        ..dsr = SerialPortDsr.flowControl
        ..dtr = SerialPortDtr.flowControl
        ..setFlowControl(SerialPortFlowControl.rtsCts);
          print("Porta aberta com sucesso!");
          Varcontroller.instance.statusESP = true;
          Varcontroller.instance.estado = "Conectado";
          Varcontroller.instance.reader?.stream.listen((data) {

            String received = String.fromCharCodes(data);
            _buffer += received;

            if (_buffer.contains('\n')) {
              print(_buffer);
              final dados = _buffer;
              Varcontroller.instance.bits = dados;
              _buffer = "";
            }
          });
        }