jpnurmi/flutter_libserialport #68: Example on Ubuntu 22.10 gives `errno = 22`

Schwusch, 2023-01-03:
Running the example on Ubuntu 22.10 with Flutter `3.7.0-1.2.pre` gives this error on startup:

```
══╡ EXCEPTION CAUGHT BY WIDGETS LIBRARY╞═══════════════════════════════════════════════════════════
The following SerialPortError was thrown building Builder(dirty):
Invalid argument, errno = 22

The relevant error-causing widget was:
  Builder
  Builder:file:///home/schwusch/development/flutter_libserialport/example/lib/main.dart:52:17

When the exception was thrown, this was the stack:
#0      Util.call (package:libserialport/src/util.dart:39:7)
#1      Util.toInt (package:libserialport/src/util.dart:77:16)
#2      _SerialPortImpl.busNumber (package:libserialport/src/port.dart:284:17)
#3      _ExampleAppState.build.<anonymous closure>
(package:flutter_libserialport_example/main.dart:59:52)
#4      Builder.build (package:flutter/src/widgets/basic.dart:7448:48)
```

Running `flutter --version`:
```
Flutter 3.7.0-1.2.pre • channel beta • https://github.com/flutter/flutter.git
Framework • revision c29b09b878 (2 weeks ago) • 2022-12-19 18:30:12 -0600
Engine • revision 8c99b0feaf
Tools • Dart 2.19.0 (build 2.19.0-444.2.beta) • DevTools 2.20.0
```

Schwusch, 2023-01-03:
When skimming through old issues, I saw the trick with `LIBSERIALPORT_DEBUG=1`, and this is the output:
```
sp: sp_get_port_by_name(/dev/ttyS4, 0x7f48e80c4990) called.
sp: Building structure for port /dev/ttyS4.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name returning SP_OK.
sp: sp_get_port_description(0x7f48e847dde0) called.
sp: sp_get_port_description returning ttyS4.
sp: sp_get_port_transport(0x7f48e847dde0) called.
sp: sp_get_port_transport returning 0.
sp: sp_get_port_usb_bus_address(0x7f48e847dde0) called.
sp: sp_get_port_usb_bus_address returning SP_ERR_ARG: Port does not use USB transport.
sp: sp_last_error_message() called.
sp: sp_last_error_message returning Invalid argument.
sp: sp_free_error_message(Invalid argument) called.
sp: sp_free_error_message returning.
sp: sp_last_error_code() called.
sp: sp_last_error_code returning 22.
sp: sp_last_error_message() called.
sp: sp_last_error_message returning Invalid argument.
sp: sp_free_error_message(Invalid argument) called.
sp: sp_free_error_message returning.
sp: sp_last_error_code() called.
sp: sp_last_error_code returning 22.
```

mateen-demah, 2023-05-20:
@Schwusch did you find a fix or workaround for this error? Facing the problem here.

Schwusch, 2023-05-20:
@mateen-demah I did port the [Golang serial library](https://github.com/bugst/go-serial/blob/v1.5.0/serial_unix.go#L292) function for getting a more valid list of ports, but abandoned that code in favour of just filtering on vendor/product ID.
I seriously considered porting all of Golangs serial library in order to have a pure Dart implementation (no `libserialport` dependency).
Maybe at some point in the future.

mateen-demah, 2023-05-20:
> @mateen-demah I did port the [Golang serial library](https://github.com/bugst/go-serial/blob/v1.5.0/serial_unix.go#L292) function for getting a more valid list of ports, but abandoned that code in [...]

@Schwusch how did you do that? From the example app, you'd have create a SerialPort object to get access to the vendor/product ID. And the error occurs at the point of SerialPort object creation.

Schwusch, 2023-05-20:
My use case was to present the user with a list of available ports to choose from, and that is possible without instantiating a `SerialPort` object, by enumerating the files in the `/dev` directory like they do it in `go-serial`.
After that, you can instantiate the object with the chosen name `SerialPort(name);`.
Also, it is a good idea to wrap the instantiation with a `try-catch` :)

Schwusch, 2023-05-20:
Also, you can read the [vid/pid from the file system](https://github.com/bugst/go-serial/blob/v1.5.0/enumerator/usb_linux.go#L36)

mateen-demah, 2023-05-20:
Thanks @Schwusch I'll try that out. In my case, I don't just want to present the ports so I really need that object.

mingpepe, 2023-05-23:
> When skimming through old issues, I saw the trick with `LIBSERIALPORT_DEBUG=1`, and this is the output:
>
> ```
> sp: sp_get_port_by_name(/dev/ttyS4, 0x7f48e80c4990) called.
> sp: Building structure for [...]

Seems like your serial port is not USB type, hence no bus number can be provided.
Possible throw an exception from [here](https://github.com/jpnurmi/flutter_libserialport/blob/main/example/lib/main.dart#L59), then trace code can find that the library just complain about it's type when getting bus number.
https://github.com/jpnurmi/libserialport.dart/blob/main/lib/src/port.dart#L285
https://github.com/martinling/libserialport/blob/master/serialport.c#L172

mateen-demah, 2023-05-25:
Thanks @mingpepe . For my usecase I'm only interested in USB devices so I put a try block around the port creation code and just skipped when something went wrong. Worked like a charm; all the USB devices showed up.