jpnurmi/flutter_libserialport #71: Invalid argument(s): length must be in the range [0, 4611..]

MervePlt, 2023-04-04:
It does not show this error all the time, but when it does, the communication can not be established by this port again. But, turning it off and on manually fixes the problem. However, we can not count that as a constant solution for this problem. Is there any other way we can follow to solve it automatically? (without turn the port on and off MANUALLY) !!

[image]

Blazym8, 2023-04-20:
Same, i assume its the buffer on win10 that gets fuller and fuller until you invoke SerialPortReader where id assume everything gets dumped at once into the stream.

Unfortunately flushing or draining the selected port before reopening the reader didn't help either...

_Note:_
This issue seems to be windows-related since it never happened on macos for me.

joaquin-amenta, 2023-05-26:
Same here, have you found a solution?

mingpepe, 2023-05-29:
`sp_nonblocking_read` may [return negtive value on error](https://github.com/jpnurmi/libserialport.dart/blob/main/lib/src/bindings.dart#L1459). This library assume that read operation is [always success](https://github.com/jpnurmi/libserialport.dart/blob/main/lib/src/util.dart#L48). If the return value is negative, `asTypedList` may throw that exception.

You can use `sp_last_error_message` to check what error occured.

Ido-Sobol, 2023-06-11:
> `sp_nonblocking_read` may [return negtive value on error](https://github.com/jpnurmi/libserialport.dart/blob/main/lib/src/bindings.dart#L1459). This library assume that read operation is [always [...]

How do I call `sp_last_error_message` when I'm using flutter_libserialport and not libserialport?

mingpepe, 2023-06-12:
https://github.com/jpnurmi/libserialport.dart/blob/main/lib/src/bindings.dart#L1857

samuelishimwe, 2024-03-21:
@lucafabbri Am currently on 0.4.0 but am facing the same issue

MironRobert, 2024-08-30:
Has anyone found something about this issue? I have the same issue but not at the same place. Because for me, it happens only when I manually turn it OFF and ON.

The only thing I know is that, just disconnecting the port gets an "errno = 5" (and can sometimes crash the app). Yet reconnection automatically gets "errno = 0".

But to get that i just printed ("SerialPort.lastError") and the Unhandled exception wasn't there anymore. Just the errors listed before.

And i just don't understand why at some point the errno = 5 crashes my app.

I'm in 0.4.0.

jsalvini, 2024-09-13:
I am also on version 0.4.0 and facing the same problem, any solution to this?