jpnurmi/flutter_libserialport #74: How to listen to serial port connectivity continuously

hosseinvejdani, 2023-04-23:
How to listen to serial port connectivity continuously in Flutter Linux app with flutter_libserialport package?

realrk95, 2023-04-24:
Listen to SerialPortEvents. Check what is the output. Typically it is Event type which can equate to bool indicating the connectivity status. Put it in the initState of the connection screen and listen to this through a changenotifier to switch states when the device disconnects/connects. Looks something like this:

```
SerialPortEvents.listen((status) {
    if (event == true)
        connection procedure;
    else disconnect procedure;
})
```

hosseinvejdani, 2023-05-18:
> 

But there is no `SerialPortEvents` class defined in `flutter_libserialport` package!

mateen-demah, 2023-06-17:
@hosseinvejdani have you found a way to this?

hosseinvejdani, 2023-06-17:
> @hosseinvejdani have you found a way to this?

No!

ohdonpiano, 2023-06-19:
something like this?
https://pub.dev/documentation/dart_serial_port/latest/dart_serial_port/SerialPortReader-class.html

an example:

void createReader(SerialPort serialPort) {
    final portReader = SerialPortReader(serialPort);
    readerSubscription = portReader.stream.listen((newData) {
      log("new data received from serial port: ${newData.length} bytes");
    });
  }

  void dispose() {
    readerSubscription?.cancel();
  }

mateen-demah, 2023-06-19:
> something like this? https://pub.dev/documentation/dart_serial_port/latest/dart_serial_port/SerialPortReader-class.html
>
> an example:
>
> ```
> void createReader(SerialPort serialPort) {
>     final [...]

Your example is for getting the data stream from the port. This issue is about getting the status of the connection on the port. for instance, if a USB device is suddenly disconnected, how do you detect that in the application?

ohdonpiano, 2023-06-19:
I don't see this feature on the serial port directly

reading SerialPortReader code, I find this part:

```
_receiver = ReceivePort();
    _receiver!.listen((data) {
      if (data is SerialPortError) {
        _controller.addError(data);
      } else if (data is Uint8List) {
        _controller.add(data);
      }
    });
```

maybe you can try something like this:

```
readerSubscription = portReader.stream.listen((newData) {
      log("new data received from serial port: ${newData.length} bytes");
      commandReceived(newData);
    }, onError: (error) {
      if (error is SerialPortError) {
        log("received serial port error: $error");
        dispose();
      }
    });
```

I can't try it myself yet

mateen-demah, 2023-06-19:
Alright. 
Thanks @ohdonpiano , maybe I should take a look all the possible SerialPortErrors to find one I can look out for.

oteriusus, 2023-12-18:
> I don't see this feature on the serial port directly
>
> reading SerialPortReader code, I find this part:
>
> ```
> _receiver = ReceivePort();
>     _receiver!.listen((data) {
>       if (data is [...]

Thanks @ohdonpiano, it works for me!