jpnurmi/flutter_libserialport #75: Getting wrong data format

lar-asmaoui, 2023-04-25:
Hi,

import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_libserialport/flutter_libserialport.dart';

class SerialPorts extends ChangeNotifier {
  List<String> _ports = SerialPort.availablePorts;
  String _res = "";
  SerialPort? printerPort;
  SerialPort? scalePort;

  List<String> get availablePorts => _ports;

  double get getData => double.parse(_res);
  set setData(String data) {
    _res = data;
    notifyListeners();
  }

  void setPortName(String name) {}

  void listAllPorts() {
    _ports = SerialPort.availablePorts;
    notifyListeners();
  }

  Future<void> connect(String portName) async {
    try {
      printerPort = SerialPort(portName);
      printerPort!.open(mode: 1);
      printerPort!.config = SerialPortConfig()
        ..baudRate = 9600
        ..bits = 8
        ..stopBits = 1
        ..parity = SerialPortParity.none
        ..rts = SerialPortRts.flowControl
        ..cts = SerialPortCts.flowControl
        ..dsr = SerialPortDsr.flowControl
        ..dtr = SerialPortDtr.flowControl
        ..setFlowControl(SerialPortFlowControl.rtsCts);
      if (printerPort!.isOpen) {
        read();
        notifyListeners();
      }
    } catch (e) {
      print(e.toString());
      printerPort!.close();
      notifyListeners();
    }
  }

  Future<void> read() async {
    try {
      SerialPortReader reader = SerialPortReader(printerPort!, timeout: 3000);
      Stream<String> upcomingData = reader.stream.map(
        (data) => String.fromCharCodes(data).replaceAll('\r\n', ''),
      );
      upcomingData.listen((data) {
        _res += data.replaceAll('\r\n', '');
        print(_res);
      });
      notifyListeners();
    } on SerialPortError catch (err, _) {
      if (printerPort!.isOpen) {
        printerPort!.close();
        print(SerialPort.lastError);
        notifyListeners();
      }
    }
  }

  void disconnect() {
    printerPort!.dispose();
    scalePort!.dispose();
    printerPort!.close();
    scalePort!.close();
    print("disconnected !!");
    notifyListeners();
  }
}

I run this code and I got data like this : 
28.0028.0028.0028.0028.0028.00

I want to print each value per each line.please help.

realrk95, 2023-04-26:
What is the delimiter or indication of new line? I would recommend capturing the values and using regex patterns to do this.

lar-asmaoui, 2023-04-26:
> What is the delimiter or indication of new line? I would recommend capturing the values and using regex patterns to do this.

I think the error goes here:
Stream upcomingData = reader.stream.map(
(data) => String.fromCharCodes(data).replaceAll('\r\n', ''),
);
upcomingData.listen((data) {
_res += data.replaceAll('\r\n', '');
print(_res);
});

closed 2024-02-21 by lucafabbri