jpnurmi/flutter_libserialport #77: Can't read String data with low BaudRate

nilotpalkapri, 2023-05-21:
I am trying to read stream data from Arduino UNO with 9600 baudrate. Problem is, it is not reading the whole string at a time.
Elaboration: Suppose I want to read repeatedly "Hello, I am Nilotpal." It is giving output like "Hel", new line "lo," again new line and so on.
Code:
Stream<String> upcomingData = reader.stream.map((data) => String.fromCharCodes(data));
upcomingData.listen((data){print(data);});

Note: It is working with higher baudrate like 2000000. 

Is there any solution for this?

nilotpalkapri, 2023-05-24:
So, I found a workaround by adding a delimitator at the end of the data sending from arduino, and in the application concatenating data till the delimitator received & spliting using the same.

closed 2023-05-24 by nilotpalkapri

MihirGit1, 2023-08-21:
> So, I found a workaround by adding a delimitator at the end of the data sending from arduino, and in the application concatenating data till the delimitator received & spliting using the same.

Not a workaround, that's a standard practice. Devices communicating through serial ports have been using stop bits (bits that indicate end of data) for ages now!

MihirGit1, 2023-08-22:
> > > So, I found a workaround by adding a delimitator at the end of the data sending from arduino, and in the application concatenating data till the delimitator received & spliting using the same.
> > 
> > 
> > Not a workaround, that's a standard practice. Devices communicating through serial ports have been using stop bits (bits that indicate end of data) for ages now!
> 
> I don't think you have ever used serial port with flutter. Python does read full line from serial port at once but in flutter with this package we can't. So, anyhow I made a temporary solution. If you think it is not a workaround, please provide one without writing unnecessary comments.

My experience with serial communication goes way back, in fact my programming journey itself started with serial communication on .net apps. Furthermore, I am actively using flutter_libserialport. I think you got the wrong idea though, what I meant to say is that what you stumbled across is more of a solution than a workaround. Anyway, more info here - https://www.electronicwings.com/raspberry-pi/raspberry-pi-uart-communication-using-python-and-c , have a good day.