jpnurmi/flutter_libserialport #81: android build failed

toplinuxsir, 2023-06-30:
`
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/2j5c5l25/arm64-v8a'
  [1/2] Building C object CMakeFiles/serialport.dir/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o
  FAILED: CMakeFiles/serialport.dir/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o 
  /data/android_sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-none-linux-android21 --gcc-toolchain=/data/android_sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/data/android_sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport -I/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -fno-limit-debug-info  -fPIC -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT CMakeFiles/serialport.dir/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -MF CMakeFiles/serialport.dir/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o.d -o CMakeFiles/serialport.dir/home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -c /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:9: error: invalid application of 'sizeof' to an incomplete type 'struct termiox'
          return sizeof(struct termiox);
                 ^     ~~~~~~~~~~~~~~~~
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:23: note: forward declaration of 'struct termiox'
          return sizeof(struct termiox);
                               ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:15: error: incomplete definition of type 'struct termiox'
          *rts = (termx->x_cflag & RTSXOFF);
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:27: error: use of undeclared identifier 'RTSXOFF'
          *rts = (termx->x_cflag & RTSXOFF);
                                   ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:15: error: incomplete definition of type 'struct termiox'
          *cts = (termx->x_cflag & CTSXON);
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'CTSXON'
          *cts = (termx->x_cflag & CTSXON);
                                   ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:15: error: incomplete definition of type 'struct termiox'
          *dtr = (termx->x_cflag & DTRXOFF);
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:27: error: use of undeclared identifier 'DTRXOFF'
          *dtr = (termx->x_cflag & DTRXOFF);
                                   ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:15: error: incomplete definition of type 'struct termiox'
          *dsr = (termx->x_cflag & DSRXON);
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:27: error: use of undeclared identifier 'DSRXON'
          *dsr = (termx->x_cflag & DSRXON);
                                   ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:7: error: incomplete definition of type 'struct termiox'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
          ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:22: error: use of undeclared identifier 'RTSXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                              ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:32: error: use of undeclared identifier 'CTSXON'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                        ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:41: error: use of undeclared identifier 'DTRXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:51: error: use of undeclared identifier 'DSRXON'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                                           ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:121:8: error: incomplete definition of type 'struct termiox'
[... 10 lines cut ...]
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:123:21: error: use of undeclared identifier 'CTSXON'
                  termx->x_cflag |= CTSXON;
                                    ^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:125:8: error: incomplete definition of type 'struct termiox'
                  termx->x_cflag |= DTRXOFF;
                  ~~~~~^
  /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  20 errors generated.
  ninja: build stopped: subcommand failed.

  C++ build system [build] failed while executing:
      /data/android_sdk/cmake/3.18.1/bin/ninja \
        -C \
        /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/2j5c5l25/arm64-v8a \
        serialport
    from /home/lhs/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android

`

chasersmith, 2023-07-17:
+1 having same problem.

ajeeshklr, 2023-07-19:
+1 Facing same issue.

doubletcjs, 2023-08-18:
Same issue.

It work for me like this.

Before:
[image]
After:
[image]

Exevil33, 2023-08-24:
I've tried as you have done, but i doesn't work for me, same error..

[image]

and the error:

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.
> Build command failed.
  Error while executing process /home/exevil32/Android/Sdk/cmake/3.18.1/bin/ninja with arguments {-C /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/3962t4v3/arm64-v8a serialport}
  ninja: Entering directory `/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/3962t4v3/arm64-v8a'
  [1/2] Building C object CMakeFiles/serialport.dir/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o
  FAILED: CMakeFiles/serialport.dir/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o 
  /home/exevil32/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-none-linux-android21 --gcc-toolchain=/home/exevil32/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/exevil32/Android/Sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/linux-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport -I/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -fPIC -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT CMakeFiles/serialport.dir/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -MF CMakeFiles/serialport.dir/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o.d -o CMakeFiles/serialport.dir/home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -c /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:43:9: error: use of undeclared identifier 'TCGETS2'
          return TCGETS2;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:52:9: error: use of undeclared identifier 'TCSETS2'
          return TCSETS2;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:61:9: error: invalid application of 'sizeof' to an incomplete type 'struct termios2'
          return sizeof(struct termios2);
                 ^     ~~~~~~~~~~~~~~~~~
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:61:23: note: forward declaration of 'struct termios2'
          return sizeof(struct termios2);
                               ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:75:10: error: incomplete definition of type 'struct termios2'
          if (term->c_ispeed != term->c_ospeed)
              ~~~~^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:71:9: note: forward declaration of 'struct termios2'
          struct termios2 *term = (struct termios2 *)data;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:75:28: error: incomplete definition of type 'struct termios2'
          if (term->c_ispeed != term->c_ospeed)
                                ~~~~^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:71:9: note: forward declaration of 'struct termios2'
          struct termios2 *term = (struct termios2 *)data;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:78:14: error: incomplete definition of type 'struct termios2'
                  return term->c_ispeed;
                         ~~~~^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:71:9: note: forward declaration of 'struct termios2'
          struct termios2 *term = (struct termios2 *)data;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:88:6: error: incomplete definition of type 'struct termios2'
          term->c_cflag &= ~CBAUD;
          ~~~~^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:84:9: note: forward declaration of 'struct termios2'
          struct termios2 *term = (struct termios2 *)data;
                 ^
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:88:20: error: use of undeclared identifier 'CBAUD'
          term->c_cflag &= ~CBAUD;
                            ^
[... cut ...]
  /home/exevil32/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'DTRXOFF'
          *dtr = (termx->x_cflag & DTRXOFF);
                                   ^
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  20 errors generated.
  ninja: build stopped: subcommand failed.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 22s

pro100andrey, 2023-08-28:
Hi All, 
https://github.com/jpnurmi/flutter_libserialport/pull/72
https://github.com/jpnurmi/flutter_libserialport/pull/84

rekire, 2023-09-27:
Just FYI don't add "+1" comments here use the thumb up emoji instead. I would guess that just as mentioned by @pro100andrey that the PR #84 will fix that.

Without having that merged you can use that fork by adding this to your pubspec.yaml:

```yaml
  flutter_libserialport: # ^0.3.1 <- uncomment this after the 0.3.1 release of flutter_libserialport
    git: # remove this and the next two lines after the release above is live
      url: https://github.com/ajeeshklr/flutter_libserialport
      ref: 849b23ac3228ce46a22b24009becf34765494403
```

With that overwritten dependency it works on my machine.

ivanix9325, 2023-09-27:
the solution provided by @doubletcjs worked for me. After the changes he suggested I performed a Flutter Clean command and Flutter Create .

I succesfully compiled on android. I still have to test on serial hardware to validate it works 100%. I will update this comment on the result test.

yaowei103, 2023-10-28:
@doubletcjs ，works for me,  install CMake, Nanja in my PC(add Nanja exe file to CMake bin folder), then add NDK in android studio,  it works

closed 2024-02-21 by lucafabbri