jpnurmi/flutter_libserialport #83: string.h file not found

czrobi, 2023-07-20:
Hi folks, I have tried to use the package on Ubuntu 22.04.2 LTS with 5.19 and 5.11 kernels, but the building is failed. Target platform would be Linux. I got the following errors:

/home/robert/Desktop/flutter/proba2/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:37:10: fatal error: 'stdlib.h' file not found
/home/robert/Desktop/flutter/proba2/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
/home/robert/Desktop/flutter/proba2/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
/home/robert/Desktop/flutter/proba2/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
Exception: Build process failed

The string.h file is really missing, I have checked it. Anybody knows the solution for this?

lucafabbri, 2024-02-21:
hi @czrobi this should be fixed in 0.4.0, could you check if it is working now?

ChrisDeardeuff, 2024-10-19:
Hi @lucafabbri, I have version 0.4.0 and am encountering the same error:

`Launching lib/main.dart on Linux in debug mode...
Building Linux application...
/home/deck/StudioProjects/Nova/nova/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/third_party/libserialport/linux_termios.c:37:10: fatal error: 'stdlib.h' file not found
/home/deck/StudioProjects/Nova/nova/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
/home/deck/StudioProjects/Nova/nova/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
/home/deck/StudioProjects/Nova/nova/linux/flutter/ephemeral/.plugin_symlinks/flutter_libserialport/linux/libserialport/../../third_party/libserialport/libserialport_internal.h:52:10: fatal error: 'string.h' file not found
Error: Build process failed`

is there something else I am missing on my side? Thanks!