jpnurmi/flutter_libserialport #86: SerialPortError: Try again, errno = 11, Try again

qinrunhao, 2023-08-18:
I used Android Studio to connect the device to install the app, and the serial port connection was successful. However, applications installed with the packaged apk report errors in the title. I can't see the log. Please look at it.

qinrunhao, 2023-09-03:
I got the error log
```
2023-09-03 10:41:24.633 19760-19795 flutter                 com.bsmart.bsmart_pad                I  SerialPortError: Try again, errno = 11
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #0      Util.call (package:libserialport/src/util.dart:40)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #1      _SerialPortImpl.config= (package:libserialport/src/port.dart:347)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #2      SerialPortTool._buildConfig (package:bsmart_pad/utils/serial_port_tools.dart:55)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #3      SerialPortTool.start (package:bsmart_pad/utils/serial_port_tools.dart:33)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #4      _HomePageState.startSerialPort (package:bsmart_pad/pages/pad/home.dart:48)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #5      _HomePageState.initState.<anonymous closure> (package:bsmart_pad/pages/pad/home.dart:41)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #6      SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1284)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #7      SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1223)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #8      SchedulerBinding.scheduleWarmUpFrame.<anonymous closure> (package:flutter/src/scheduler/binding.dart:939)
2023-09-03 10:41:24.636 19760-19795 flutter                 com.bsmart.bsmart_pad                I  #12     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189)
```
The source code is as follows
```
static _buildConfig(int baudRate, int dataBits, int parity, int stopBits) {
    var config = SerialPortConfig();
    config.bits = dataBits;
    config.baudRate = baudRate;
    config.parity = parity;
    config.stopBits = stopBits;
    port.config = config;  // This line reported an error
  }
```

Farzad7m, 2023-09-20:
you dont open a serial port before send a data

Farzad7m, 2023-10-02:
after open serial Port, config that
like this
      serialPort.openReadWrite();
      serialPort.config = config;

njxujunjie, 2024-07-21:
Has the issue with errno=11 been resolved?