jpnurmi/flutter_libserialport #87: my code only gets the expected data after starting and closing party, it seems it the serial port object does not use the configs even setting them

geoffrey-oongo, 2023-09-06:
void getDataFromLibSerial(
      LibSerial.SerialPort? port, LibSerial.SerialPortReader? reader) async {
    try {
      if (reader != null) {
        Stream<String> upcomingData = reader.stream.map((event) {
          print(event);
          var data = ascii.decode(event);
          print(data);

          try {
            if (data.contains('S') &&
                data.contains('+') &&
                data.contains('kg')) {
              cleanedData = data
                  .split('\n')[0]
                  .split("+")[1]
                  .replaceAll(',kg', '')
                  .trim();

              var parsedWeight = num.tryParse(cleanedData);
              print(parsedWeight);
              if (parsedWeight != null) {
                cleanWeight = parsedWeight.toString();
                widget.func(cleanWeight ?? "", context);
              }
            }
          } catch (e) {}
          print(cleanWeight);

          return cleanWeight ?? "";
        });

        upcomingData.listen((event) {
          if (event.isNotEmpty) {
            setState(() {});
          }
        });
      }
    } catch (e) {
      print(e);
    }
  }

void didChangeDependencies() async {
    final connectionProvider =
        Provider.of<AuthProvider>(context, listen: false).connection;

    showWeight = true;

    if (connectionProvider != null) {
      if (connectionProvider.type == 'Serial') {
        try {
          String COM_PORT = connectionProvider.COM_PORT ?? "COM3";
          port = LibSerial.SerialPort(COM_PORT);

          if (port == null) {
            return;
          }
          LibSerial.SerialPortConfig config = LibSerial.SerialPortConfig();
          config.bits = 8;
          config.baudRate = 9600;
          config.parity = 0;
          config.stopBits = 1;
          port!.config = config;
          port!.openRead();

          // _getData(context);
          reader = LibSerial.SerialPortReader(port!);

          getDataFromLibSerial(port, reader);
        } catch (e) {}
      } else if (connectionProvider.type == 'TCP/IP') {
        try {
          socket = await HttpSocketClient.http_socket(
              connectionProvider.ip_address, connectionProvider.port_number);
          _getDataFromSocket(socket);
        } catch (e) {
          print('');
        }
      }
    }
    super.didChangeDependencies();
  }

lucafabbri, 2024-02-21:
@geoffrey-oongo check the #29 let me know if it solves the issue

geoffrey-oongo, 2024-04-16:
Thanks for the assistance Nataraja

suvarna84, 2024-12-13:
I am also facing same issue, if i open and closing serial moniter data is streaming otherwise it doesnot. I have configured the baud rate after opening the port. Please help to resolve this

rafaelntg, 2025-05-23:
Hi! I’m experiencing the same issue, but only when using a USB-to-RS232 converter cable (such as Prolific).

If I simply open and close the serial port (on Windows), everything works fine.

Does anyone have any ideas on how to fix this?