jpnurmi/flutter_libserialport #88: Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.

SherazAsghar37, 2023-09-12:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Users\SHERAZ\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_libserialport-0.3.0\android\.cxx\Debug\r1i3y4w6\arm64-v8a'
  [1/5] Building C object CMakeFiles/serialport.dir/afa35b95405eb662a66b79b7fc88ba64/third_party/libserialport/linux_termios.c.o
  FAILED: CMakeFiles/serialport.dir/afa35b95405eb662a66b79b7fc88ba64/third_party/libserialport/linux_termios.c.o 
  C:\Users\SHERAZ\AppData\Local\Android\Sdk\ndk\23.1.7779620\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe --target=aarch64-none-linux-android21 --gcc-toolchain=C:/Users/SHERAZ/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/SHERAZ/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -IC:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport -IC:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -fno-limit-debug-info  -fPIC -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT CMakeFiles/serialport.dir/afa35b95405eb662a66b79b7fc88ba64/third_party/libserialport/linux_termios.c.o -MF CMakeFiles\serialport.dir\afa35b95405eb662a66b79b7fc88ba64\third_party\libserialport\linux_termios.c.o.d -o CMakeFiles/serialport.dir/afa35b95405eb662a66b79b7fc88ba64/third_party/libserialport/linux_termios.c.o -c C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:9: error: invalid application of 'sizeof' to an incomplete type 'struct termiox'
          return sizeof(struct termiox);
                 ^     ~~~~~~~~~~~~~~~~
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:23: note: forward declaration of 'struct termiox'
          return sizeof(struct termiox);
                               ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:15: error: incomplete definition of type 'struct termiox'
          *rts = (termx->x_cflag & RTSXOFF);
                  ~~~~~^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:27: error: use of undeclared identifier 'RTSXOFF'
          *rts = (termx->x_cflag & RTSXOFF);
                                   ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:15: error: incomplete definition of type 'struct termiox'
          *cts = (termx->x_cflag & CTSXON);
                  ~~~~~^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'CTSXON'
          *cts = (termx->x_cflag & CTSXON);
                                   ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:15: error: incomplete definition of type 'struct termiox'
          *dtr = (termx->x_cflag & DTRXOFF);
                  ~~~~~^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:27: error: use of undeclared identifier 'DTRXOFF'
          *dtr = (termx->x_cflag & DTRXOFF);
                                   ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:15: error: incomplete definition of type 'struct termiox'
          *dsr = (termx->x_cflag & DSRXON);
                  ~~~~~^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:27: error: use of undeclared identifier 'DSRXON'
          *dsr = (termx->x_cflag & DSRXON);
                                   ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:7: error: incomplete definition of type 'struct termiox'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
          ~~~~~^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:22: error: use of undeclared identifier 'RTSXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                              ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:32: error: use of undeclared identifier 'CTSXON'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                        ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:41: error: use of undeclared identifier 'DTRXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                                 ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:51: error: use of undeclared identifier 'DSRXON'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                                           ^
  C:/Users/SHERAZ/AppData/Local/Pub/Cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:121:8: error: incomplete definition of type 'struct termiox'
                  termx->x_cflag |= RTSXOFF;
[... 22 lines cut ...]
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  20 errors generated.
  [2/5] Building C object CMakeFiles/serialport.dir/e63de0ff05a26f9f0194e98a55097f85/flutter_libserialport-0.3.0/third_party/libserialport/linux.c.o
  [3/5] Building C object CMakeFiles/serialport.dir/e63de0ff05a26f9f0194e98a55097f85/flutter_libserialport-0.3.0/third_party/libserialport/serialport.c.o
  ninja: build stopped: subcommand failed.

  C++ build system [build] failed while executing:
      @echo off
      "C:\\Users\\SHERAZ\\AppData\\Local\\Android\\sdk\\cmake\\3.18.1\\bin\\ninja.exe" ^
        -C ^
        "C:\\Users\\SHERAZ\\AppData\\Local\\Pub\\Cache\\hosted\\pub.dev\\flutter_libserialport-0.3.0\\android\\.cxx\\Debug\\r1i3y4w6\\arm64-v8a" ^
        serialport
    from C:\Users\SHERAZ\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_libserialport-0.3.0\android

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3m 28s
Exception: Gradle task assembleDebug failed with exit code 1

Getting this error after adding this package in my pubspec.yaml and running the project.

NathanTPerkins, 2023-09-14:
I am also having this issue.

Farzad7m, 2023-09-18:
This is a common problem
please fix it

gabboraron, 2023-09-19:
Hi, I got this on Zorin OS, not on windows: 

```
Launching lib/main.dart on sdk gphone x86 64 in debug mode...
Running Gradle task 'assembleDebug'...

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/h6o356w5/arm64-v8a'
  [1/2] Building C object CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o
  FAILED: CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o 
  /home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-none-linux-android21 --gcc-toolchain=/home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport -I/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -fno-limit-debug-info  -fPIC -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -MF CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o.d -o CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -c /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:9: error: invalid application of 'sizeof' to an incomplete type 'struct termiox'
          return sizeof(struct termiox);
                 ^     ~~~~~~~~~~~~~~~~
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:23: note: forward declaration of 'struct termiox'
          return sizeof(struct termiox);
                               ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:15: error: incomplete definition of type 'struct termiox'
          *rts = (termx->x_cflag & RTSXOFF);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:27: error: use of undeclared identifier 'RTSXOFF'
          *rts = (termx->x_cflag & RTSXOFF);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:15: error: incomplete definition of type 'struct termiox'
          *cts = (termx->x_cflag & CTSXON);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'CTSXON'
          *cts = (termx->x_cflag & CTSXON);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:15: error: incomplete definition of type 'struct termiox'
          *dtr = (termx->x_cflag & DTRXOFF);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:27: error: use of undeclared identifier 'DTRXOFF'
          *dtr = (termx->x_cflag & DTRXOFF);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:15: error: incomplete definition of type 'struct termiox'
          *dsr = (termx->x_cflag & DSRXON);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
[... cut ...]
        /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/h6o356w5/arm64-v8a \
        serialport
    from /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 19s
Exception: Gradle task assembleDebug failed with exit code 1
```

SherazAsghar37, 2023-09-21:
I was working with Modbus serial communication and was using modbus_client_serial, this package is dependent on flutter_libserialport.
I was running my application on emulator/real android device and was facing this issue, then I installed Microsoft visual studio and install C++ desktop development then I ran my application on windows and the error was gone. 
I don't know how it got solved but what I thought was serial communication is not possible with phone as I was unable to find any android application, in the mean time my windows software was doing well with serial so I tried this approach and it worked.
correct me if serial communication is possible with flutter android application.

gabboraron, 2023-09-25:
@SherazAsghar37  There are several flutter packages which are saying that they are able to do it, but they say only on android. This one says that this is able to handle on Android, Linux, Windows. BTW on Linux you need to give permission because the serial communication is a security question, so you have to give the permission to the app if you want to run it.

[image]

gabboraron, 2023-09-26:
@SherazAsghar37 Tried what you said, in VS code on Windows, an it not able to sand anything with serial...

SherazAsghar37, 2023-09-26:
@gabboraron I've tried a lot with my windows technique but also unable to exchange any data. I was working with virtual com port and was trying to do Modbus serial communication and at the end I shut It down.
Secondly this package claims about android but I think there are some errors with the package's android functionality.

SherazAsghar37, 2023-09-26:
I've tried several Packages but didn't found the appropriate one yet

dwikyintimap, 2023-10-02:
i got the same error on MacOs, anyone know the alternative package for flutter serial?

chehansivarubanvega, 2023-12-20:
I am facing this issue. Please resolve this

JuliadsonS, 2023-12-24:
/Users//.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifies this file on line 57 to #undef HAVE_STRUCT_TERMIOX

ShengjieGeng, 2024-01-09:
> /Users//.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifies this file on line 57 to #undef HAVE_STRUCT_TERMIOX

Awesome!!! it worked. Thanks a lot.

unleed-l, 2024-01-23:
> /Users//.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifies this file on line 57 to #undef HAVE_STRUCT_TERMIOX

Can't find this file on mac, any tip?

JuliadsonS, 2024-01-24:
> > /Users//.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifica este arquivo na linha 57 para #undef HAVE_STRUCT_TERMIOX
> 
> Não consigo encontrar este arquivo no mac, alguma dica?

Users/<nomedoseusuario>/..... faz uma buscar pelo finder

unleed-l, 2024-01-24:
> > > /Users//.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifica este arquivo na linha 57 para #undef HAVE_STRUCT_TERMIOX
> > 
> > 
> > Não consigo encontrar este arquivo no mac, alguma dica?
> 
> Users/nomedoseuusuario/..... faz uma buscar pelo finder
> 
> https://support.apple.com/pt-br/guide/mac-help/mchlp1236/mac

Por alguma razão a busca do finder não estava encontrando o arquivo, segui o caminho pelo terminal e consegui fazer a alteração, obrigado.

closed 2024-02-21 by lucafabbri