jpnurmi/flutter_libserialport #89: build error for ARM, can't find something for `struct termiox`

gabboraron, 2023-09-20:
Hi, I got this on Zorin OS, android studio:

```
Launching lib/main.dart on sdk gphone x86 64 in debug mode...
Running Gradle task 'assembleDebug'...

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/h6o356w5/arm64-v8a'
  [1/2] Building C object CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o
  FAILED: CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o 
  /home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang --target=aarch64-none-linux-android21 --gcc-toolchain=/home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64 --sysroot=/home/sandor/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -DLIBSERIALPORT_ATBUILD -Dserialport_EXPORTS -I/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport -I/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/../../third_party/libserialport -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -fno-limit-debug-info  -fPIC -std=c99 -Wall -Wextra -pedantic -Wmissing-prototypes -Wshadow -MD -MT CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -MF CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o.d -o CMakeFiles/serialport.dir/home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c.o -c /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:9: error: invalid application of 'sizeof' to an incomplete type 'struct termiox'
          return sizeof(struct termiox);
                 ^     ~~~~~~~~~~~~~~~~
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:98:23: note: forward declaration of 'struct termiox'
          return sizeof(struct termiox);
                               ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:15: error: incomplete definition of type 'struct termiox'
          *rts = (termx->x_cflag & RTSXOFF);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:106:27: error: use of undeclared identifier 'RTSXOFF'
          *rts = (termx->x_cflag & RTSXOFF);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:15: error: incomplete definition of type 'struct termiox'
          *cts = (termx->x_cflag & CTSXON);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:107:27: error: use of undeclared identifier 'CTSXON'
          *cts = (termx->x_cflag & CTSXON);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:15: error: incomplete definition of type 'struct termiox'
          *dtr = (termx->x_cflag & DTRXOFF);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:108:27: error: use of undeclared identifier 'DTRXOFF'
          *dtr = (termx->x_cflag & DTRXOFF);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:15: error: incomplete definition of type 'struct termiox'
          *dsr = (termx->x_cflag & DSRXON);
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:103:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:109:27: error: use of undeclared identifier 'DSRXON'
          *dsr = (termx->x_cflag & DSRXON);
                                   ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:7: error: incomplete definition of type 'struct termiox'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
          ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:22: error: use of undeclared identifier 'RTSXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                              ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:32: error: use of undeclared identifier 'CTSXON'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
                                        ^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:118:41: error: use of undeclared identifier 'DTRXOFF'
          termx->x_cflag &= ~(RTSXOFF | CTSXON | DTRXOFF | DSRXON);
[... 24 lines cut ...]
                  ~~~~~^
  /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/third_party/libserialport/linux_termios.c:116:9: note: forward declaration of 'struct termiox'
          struct termiox *termx = (struct termiox *) data;
                 ^
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  20 errors generated.
  ninja: build stopped: subcommand failed.

  C++ build system [build] failed while executing:
      /home/sandor/Android/Sdk/cmake/3.18.1/bin/ninja \
        -C \
        /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/.cxx/Debug/h6o356w5/arm64-v8a \
        serialport
    from /home/sandor/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 19s
Exception: Gradle task assembleDebug failed with exit code 1
```

Noninus, 2023-09-20:
same error here

dwikyintimap, 2023-10-02:
i got the same error, any solution?

gabboraron, 2023-10-02:
@dwikyintimap not yet

starovoid, 2023-10-30:
How do you work around this problem?

bookshiyi, 2023-12-24:
Possible useful reference information: 
https://github.com/ajeeshklr/flutter_libserialport/commit/aa5abd02ed344bf141134a99a7ddd07afab794aa

JuliadsonS, 2023-12-24:
/Users/<user>/.pub-cache/hosted/pub.dev/flutter_libserialport-0.3.0/android/libserialport/config.h modifies this file on line 57 to #undef HAVE_STRUCT_TERMIOX

closed 2024-02-21 by lucafabbri