jpnurmi/flutter_libserialport #94: How to read serial data

mihailacusteanu, 2023-10-28:
I tried multiple snippets found in the documentation and here in the issues, but no luck.

I am using a esp32 and I send on the serial port some data using this micropython code:
```
from time import sleep

while True:
    print('{"firmware_type": "keypad", "value": 2}')
    sleep(1)
```

My dart function is:

```
void readSerialData() async {
    availablePorts = SerialPort.availablePorts;
    print(availablePorts);
    SerialPort port = SerialPort("/dev/cu.wchusbserial55390119721");
    final config = SerialPortConfig();
    config.baudRate = 115200;
    port.config = config;
    SerialPortReader reader = SerialPortReader(port, timeout: 10000);

    int bufferIndex = 0;

    Stream<Uint8List> upcommingData = reader.stream.map((data) {
      return data;
    });

    try {
      port.openReadWrite();

      List<int> buffer = [];
      upcommingData.listen((data) {
        buffer.addAll(data);
        bufferIndex += data.length;

        debugPrint('Raw data: $data');

        if (bufferIndex >= 5) {
          String strBuffer = String.fromCharCodes(buffer);
          debugPrint('Buffer: $strBuffer');
          bufferIndex = 0;
          buffer = [];
        }
      });
    } on SerialPortError catch (err, _) {
      debugPrint('SerialPortError: $err');
      port.close();
    } finally {
    }
  }
```

My output is:
```
flutter: Raw data: [101, 39, 58, 32, 50, 125, 13, 10, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32, 39, 118, 97, 108, 117, 101, 39, 58, 32, 50, 125, 13, 10, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 95, 116, 121, 112, 101, 39, 58, 32, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32, 39, 118, 97, 108, 117, 101, 39, 58, 32, 50, 125, 13, 10, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 95, 116, 121, 112, 101, 39, 58, 32, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32, 39, 118, 97, 108, 117, 101, 39, 58, 32, 50, 125, 13, 10, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 95, 116, 121, 112, 101, 39, 58, 32, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32, 39, 118, 97, 108, 117, 101, 39, 58, 32, 50, 125, 13, 10, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 95, 116, 121, 112, 101, 39, 58, 32, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32, 39, 118, 97, 108, 117, 101, 39, 58, 32, 50, 125, 13, 10, 123, 39, 102, 105, 114, 109, 119, 97, 114, 101, 95, 116, 121, 112, 101, 39, 58, 32, 39, 107, 101, 121, 112, 97, 100, 39, 44, 32]
flutter: Buffer: e': 2}
flutter: 'keypad', 'value': 2}
flutter: {'firmware{'firmware_type': 'keypad', 'value': 2}
flutter: {'firmware_type': 'keypad', 'value': 2}
flutter: {'firmware_type': 'keypad', 'value': 2}
flutter: {'firmware_type': 'keypad',
lutter: {'firmware_type': 'keypad',
flutter: Raw data: [226, 161, 134]
flutter: Raw data: [196, 200, 30, 255]
flutter: Raw data: [167, 134, 158]
flutter: Raw data: [248, 240, 78, 255]
flutter: Raw data: [196, 132, 230]
flutter: Raw data: [238, 140]
flutter: Raw data: [255]
flutter: Raw data: [244, 232, 94, 255]
flutter: Raw data: [196, 132, 204]
flutter: Raw data: [242, 161, 134]
flutter: Raw data: [130, 33, 134]
flutter: Raw data: [249, 245, 66, 255]
flutter: Raw data: [196, 132, 204]
flutter: Raw data: [242, 129, 134]
flutter: Buffer: â¡ÄÈÿ§øðNÿÄæîÿôè^ÿÄÌò¡!ùõBÿÄÌò
flutter: Raw data: [238, 12, 63, 255]
flutter: Raw data: [249, 245, 66, 255]
flutter: Raw data: [74, 225, 249]
...
```
It seems its reading the right data at a certain point but then its not working.
My guess is that it's either the buffer size or the new line char '\n', but I am not sure how to tackle that.

jdekarske, 2023-11-20:
There's a bug in setting the configuration. See: https://github.com/jpnurmi/libserialport.dart/issues/24#issuecomment-812918216

mihailacusteanu, 2023-11-21:
@jdekarske you are right 😅 thanks a lot! 
This is very very helpful since I couldn't  change the default baud rate for micropython(115200) without side effects. 🥳

closed 2023-11-21 by mihailacusteanu