jpnurmi/flutter_libserialport #96: Can't read buffer that was not empty before opening serial port (Windows)

ludesert, 2023-11-12:
I'm using Windows 10 with a serial port I can manually write data to via a terminal to test the following code:
```
SerialPort port = SerialPort("COM42");

port.openReadWrite();
port.flush(); // not working ??
SerialPortReader reader = SerialPortReader(port);

reader.stream.listen((data) {
  print(data);
  port.write(data);
});
```

It works when the serial port's buffer is empty: the raw data is sent back to the serial port.

When I send 3 bytes or more via my device BEFORE starting the previous code, nothing happens: dart/flutter can't read any new byte i send via my device.
With 2 bytes before starting the code, it's ok, but if it's more than 3, the data somehow gets stuck.

I tried using the flush() and the drain() methods but none of them change anything.
If I try to print port.bytesAvailable, I get 0.

Please note that this is not related to my device as I don't have this issue while using PySerial on Python.

Any idea ?

ludesert, 2023-11-13:
Solved ... I did not configure the port correctly (at all, in fact).

Made some attempts and looks like I had to configure the dtr to 1 with libserialport while pyserial somehow did it for me.
I still don't know why not configuring the baudrate does not cause any problem, but I do it anyway.

```
SerialPortConfig conf = port.config;
confg;baudrate = 9600;
conf.dtr = 1;
port.config = conf;
```
This solved everything.

closed 2023-11-13 by ludesert

mihailacusteanu, 2023-11-13:
Does it work with 115200 baudrate?

ludesert, 2023-11-13:
If I set my serial device to 115200, yes of course.