jpnurmi/flutter_libserialport #97: How to configure baudrate and port

tkumark, 2024-01-01:
I am trying to rewrite below python code in flutter/dart. I would like to how to set port,baudrate,parity and open connection?
 
```
import time
import serial
import re

ser = serial.Serial(
    port='/dev/ttyS0',
    baudrate=115200,
    parity=serial.PARITY_NONE,
    stopbits=serial.STOPBITS_ONE,
    bytesize=serial.SEVENBITS,
)

ser.isOpen()
out = ''
for i in range(10):
    waitingQueue = ser.inWaiting()
    if waitingQueue == 0:
        weight = "-- kg"
    else:
        out = ser.readline()
        if out != '':
            print(str(out))
            m = re.search('\s\s(.+?)\s\s\s\s', str(out))
            if m != None:
                weight = m.group(1)
            else:
                weight = "Extract Error"
    print(weight.strip())

ser.close()
```

sayanp23, 2024-02-17:
Hi, Did anyone got solution for setting the Baudrate?? If you've found any solution please help me out, thanks.

lucafabbri, 2024-02-21:
Please se #29

closed 2024-02-21 by lucafabbri