jpnurmi/flutter_libserialport #100: Support for Gradle 8.3.0: Define namespace

kay4ik, 2024-03-12:
Hey,
when running an android build using the latest gradle version, following error occurs:
```console
A problem occurred configuring project ':flutter_libserialport'.
> Could not create an instance of type com.android.build.api.variant.impl.LibraryVariantBuilderImpl.
   > Namespace not specified. Specify a namespace in the module's build file. See https://d.android.com/r/tools/upgrade-assistant/set-namespace for information about setting the namespace.
      If you've specified the package attribute in the source AndroidManifest.xml, you can use the AGP Upgrade Assistant to migrate to the namespace value in the build file. Refer to https://d.android.com/r/tools/upgrade-assistant/agp-upgrade-assistant for general information about using the AGP Upgrade Assistant.
```

To solve this problem following line needs to be defined (probably):
In ```android/src/build.gradle```
```gradle
android {
    namespace "org.sigrok.flutter_libserialport"
    compileSdk 29

    kotlinOptions {
        jvmTarget = '1.8'
    }
...
```

Also the min SDK version should be increased to 19 according to this warning:
```console
C/C++: Platform version 16 is unsupported by this NDK, using 19 instead. Please change minSdk to at least 19 to avoid this warning.
```

This should be everything the console is asking for :)

## Recommendation
- I would reccomend to upgrade to flutter 3.19.x and to follow the instructions on [Deprecated imperative apply of Flutter's Gradle plugins](https://docs.flutter.dev/release/breaking-changes/flutter-gradle-plugin-apply)

---
I have avoided all warnings and errors by making following changes:
- Increasing the kotlin version to 1.9.20
- replacing ```jCentral()``` repository with ```mavenCentral()``` because it is already deprecated
- **adding the namespace**
- increasing the ```compileSdkVersion``` to 34
- adding the ```jvmTarget``` of 1.8
- increasing the ```minSdkVersion``` to 19

Now there are no warnings and errors anymore and the ```android/src/build.gradle``` is looking like that:
```gradle
group 'org.sigrok.flutter_libserialport'
version '1.0-SNAPSHOT'

buildscript {
    ext.kotlin_version = '1.9.20'
    repositories {
        google()
        mavenCentral()
    }

    dependencies {
        classpath 'com.android.tools.build:gradle:4.1.3'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
    }
}

rootProject.allprojects {
    repositories {
        google()
        mavenCentral()
    }
}

apply plugin: 'com.android.library'
apply plugin: 'kotlin-android'

android {
    namespace "org.sigrok.flutter_libserialport"
    compileSdkVersion 34

    kotlinOptions {
        jvmTarget = '1.8'
    }

    sourceSets {
        main.java.srcDirs += 'src/main/kotlin'
    }
    defaultConfig {
        minSdkVersion 19
    }
    lintOptions {
        disable 'InvalidPackage'
    }
    externalNativeBuild {
        cmake {
            path "libserialport/CMakeLists.txt"
        }
    }
}

dependencies {
    implementation "org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"
}

```

djensen47, 2024-03-28:
Any chance we can get this in a 0.5.0 or even a 1.0 release?

@kay4ik I don't supposed you forked the fix anywhere?

kay4ik, 2024-03-29:
@djensen47 No I didn't forked the solution, but I can do that next week and create a merge request. Maybe @jpnurmi can than accept this MR

ricardoboss, 2025-02-17:
Any updates on this? This is blocking us from upgrading to Flutter 3.29

kapil708, 2025-02-25:
@kay4ik  @djensen47 @ricardoboss @jpnurmi @sitic 

I am facing the same issue is there any solution?

> A problem occurred configuring project ':flutter_libserialport'.
> com.android.builder.errors.EvalIssueException: [CXX1110] Platform version 16 is unsupported by this NDK. Please change minSdk to at least 21 to avoid undefined behavior. To suppress this error, add android.ndk.suppressMinSdkVersionError=21 to the project's gradle.properties or set android.experimentalProperties["android.ndk.suppressMinSdkVersionError"]=21 in the Gradle build file.

Flutter => 3.29.0
Android SDK =>  35.0.0