jpnurmi/flutter_libserialport #102: android support

letisoft, 2024-03-26:
Hello,

sorry if am wrong, but currently there is no android support?
I had a look at the android plugin and it is empty?

Am I missing someothing?

Please let me know if you need some help, I am happy to make it work on andoird

Best regards:
V

RobologAutomation, 2024-04-01:
i to have this same doubt, is there any support for android

RobologAutomation, 2024-04-01:
im getting, serialporterror: permission denied, error 13 in android

shingohu, 2024-05-18:
> im getting, serialporterror: permission denied, error 13 in android
close selinux （adb shell setenforce 0）

rekire, 2024-06-20:
It is rarely possible to support since the permission management of android forgot to provide proper permissions to access it. On rooted devices this should be possible and maybe on special propose hardware where the manufacture had in mind to set the file permissions correctly.

curonny, 2024-07-24:
NEWS?

AurelienBallier, 2025-01-21:
Hello,
I created a PR to libserialport.dart in order to provide Android support to flutter_libserialport: 
https://github.com/jpnurmi/libserialport.dart/pull/89