jpnurmi/flutter_libserialport #104: read not working on Windows

letisoft, 2024-04-12:
Hello,
this works:
    SerialPortReader reader = SerialPortReader(_port);

    //reader.stream.
    reader.stream.listen((data) {
      print(data);
    });
this doesn't work:
Uint8List read = _port.read(1,timeout: 100000);

Of course I am not using both in the same time :)

any idea why?

Best regards:
V

letisoft, 2024-04-15:
Hello,

I am not sure this lib works at all on Windows. Does anyone ever got it working on Windows 11?

as simple as this:
    SerialPort sp = SerialPort(ports[0]);
    List<int> bytes = [0x39];
    sp.write(Uint8List.fromList(bytes));
    Uint8List read = sp.read(10);
    print(read.length);

same device is just fine with RelaTerm for example
Best regards:
V

markszente, 2024-04-17:
Not working for me either.

markszente, 2024-04-17:
It seems I can only read when I open the port using `openRead`, but not `openReadWrite`. Too bad.

micheljung, 2024-04-19:
Trying to get it work on Windows 11, too. Currently getting:

```
Invalid argument(s): length must be in the range [0, 4611686018427387903].
#0      _checkExternalTypedDataLength (dart:ffi-patch/ffi_patch.dart:62:5)
#1      Uint8Pointer.asTypedList (dart:ffi-patch/ffi_patch.dart:727:5)
#2      Util.read.<anonymous closure> (package:libserialport/src/util.dart:49:37)
#3      using (package:ffi/src/arena.dart:124:31)
#4      Util.read (package:libserialport/src/util.dart:46:16)
#5      _SerialPortReaderImpl._waitRead (package:libserialport/src/reader.dart:145:27)
#6      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:300:17)
#7      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:184:12)
```

No idea if it's me or the library but I'm using `openReadWrite` as well.

micheljung, 2024-04-20:
This solved my issue:

```
    var port = SerialPort(address);
    check(port.openReadWrite(), "Could not open for read/write: $port");
    // Don't do this
    // var config = SerialPortConfig();
    // Do those instead
    var config = port.config;
    config.baudRate = 115200;
    port.config = config;
```
Credits to https://stackoverflow.com/a/77646806/513451

closed 2024-06-17 by lucafabbri