jpnurmi/flutter_libserialport #107: Failed to load dynamic library 'libserialport.so'

TantelyLinux, 2024-06-13:
I use Elemntary OS and I have this Error : Failed to load dynamic library 'libserialport.so': dlopen failed: library "libserialport.so" not found)

TantelyLinux, 2024-06-13:
And if I re-run the project, I got this error: "error: package org.sigrok.flutter_libserialport does not exist
      flutterEngine.getPlugins().add(new org.sigrok.flutter_libserialport.FlutterLibserialportPlugin());"

lucafabbri, 2024-06-17:
Hi @TantelyLinux 

I assume Elementary OS is Linux distro and the .so library is shipped for Linux. Unfortunately I have not tested the solution on such OS and I'm aware wether Sigrok have developed or tested on such enviroment. 

Please note that this solution is a wrapper aroung libserialport.so by Sigrok

TantelyLinux, 2024-06-21:
Hi @lucafabbri 
Thanks for answering me. Now it work fine, I updated and upgraded my system than installed libserialport and remove all cache.

closed 2024-06-21 by TantelyLinux