jpnurmi/flutter_libserialport #108: Android: Namespace not specified

rekire, 2024-06-20:
I get this exception when building my app for Android:

> Failed to query the value of property 'buildFlowServiceProperty'.
> > Could not isolate value org.jetbrains.kotlin.gradle.plugin.statistics.BuildFlowService$Parameters_Decorated@17b5eafd of type BuildFlowService.Parameters
>    > A problem occurred configuring project ':flutter_libserialport'.
>       > Could not create an instance of type com.android.build.api.variant.impl.LibraryVariantBuilderImpl.
>          > Namespace not specified. Specify a namespace in the module's build file. See https://d.android.com/r/tools/upgrade-assistant/set-namespace for information about setting the namespace.
> 
>            If you've specified the package attribute in the source AndroidManifest.xml, you can use the AGP Upgrade Assistant to migrate to the namespace value in the build file. Refer to https://d.android.com/r/tools/upgrade-assistant/agp-upgrade-assistant for general information about using the AGP Upgrade Assistant.

You just need to set the namespace (and likely update some dependencies) to resolve this.

MatthewHightech, 2024-10-24:
I just came across this issue. Is there a fix that I can implement?

rekire, 2024-10-24:
Let me quote myself from the pull request with the fix.

> For others who might want to test it:
> 
> ```yaml
> dependencies:
>   // other dependencies
>   flutter_libserialport: #^0.5.0
>     # remove this git part when 0.5.0 is published
>     git:
>       url: https://github.com/snabble/flutter_libserialport.git
>       ref: 0.5.0
> ```

ImadMAKS, 2024-11-12:
Just tested it and it works with that fix!

abdulrhmanhassan, 2025-01-02:
> Let me quote myself from the pull request with the fix.

I am very grateful to you.. I have been searching for this solution for a month and a half.

PR #109 (2025-02-10): Add namespace
Fixes #108

Please note that I forked in the past a fork. Therefore the change is a bit bigger. However I see no problem when those changes go online too.

For others who might want to test it:
```yaml
dependencies:
  // other dependencies
  flutter_libserialport: #^0.5.0
    # remove this git part when 0.5.0 is published
    git:
      url: https://github.com/snabble/flutter_libserialport.git
      ref: 0.5.0
```
closed 2025-02-10 by lucafabbri