jpnurmi/flutter_libserialport #112: Android support

curonny, 2024-07-24:
Permission denied, errno = 13

rekire, 2024-12-04:
Yep, that's normal when the manufacturer forget to set correct file permissions. Basically a blind spot of Android

rekire, 2025-01-21:
Does you patch still require root or do you bypass it somehow with the other library?

AurelienBallier, 2025-01-21:
Yes you don't need root, it relies on USB Android SDK. It asks user permission upon serial port opening.

luciano-cervantes, 2025-01-28:
Hello @AurelienBallier 

I managed to advance the implementation using your PR, however, when using the `listen` method of the `SerialPortReader` class, it is returning an error:

`Fatal signal 11 (SIGSEGV), code 1, fault addr 0x420 in tid 6636 (DartWorker)`

Would you have any idea what it could be and how to fix this bug?

luciano-cervantes, 2025-01-28:
Segue meu código:

```dart
final ports = await SerialPort.availablePorts;

    StreamSubscription<Uint8List>? subscription;

    try {
      final portSerial = SerialPort(ports.first);

      final opened = await portSerial.openReadWrite();

      if (!opened) {
        return null;
      }

      portSerial.setConfig(SerialPortConfig()
        ..baudRate = 4800
        ..bits = config.bits
        ..stopBits = config.stopBits!.value
        ..parity = config.parity!.value);

      SerialPortReader serialPortReader = SerialPortReader(portSerial);

      serialPortReader.stream.listen(
        (event) {
          print(event);
        },
      );
```

AurelienBallier, 2025-01-28:
Hello @luciano-cervantes,
I haven't looked much into SerialPortReader as I only use the read/write function.
You are right it's not working as it's also tied to the underlying Sigrok libserialport, I'll have a look at it thanks for your help.
Regards,

luciano-cervantes, 2025-01-28:
@AurelienBallier 

It would be great if you could look at this, I'm in a project that needs this option, but I couldn't identify where the error is.

vinicioshenriques, 2025-02-28:
Hi @AurelienBallier, i made some changes using your branch as a base, fixing @luciano-cervantes problem, and opened a pull request, please check it out when you can.
https://github.com/AurelienBallier/libserialport.dart/pull/1